mriscoc/Ender3V2S1 #1013: [BUG] Extra character in newest release cause error

classicrocker883, 2023-08-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

```r
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void setMoveZto0()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2153:38: error: expected ';' before ')' token       
 2153 |       set_bed_leveling_enabled(false));
      |                                      ^
      |                                      ;
```

**proui/dwin.cpp** line ~2153  
`set_bed_leveling_enabled(false);`

should be
`set_bed_leveling_enabled(false));`

### Version of Professional Firmware

August 2023 (newest)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-08-06:
Thanks.
I do not recommend to have any of these options enabled: RESTORE_LEVELING_AFTER_G28, ENABLE_LEVELING_AFTER_G28
They could cause unexpected results in some calibration procedures.

mriscoc, 2023-08-17:
Fixed in latest sources.

closed 2023-08-17 by mriscoc