mriscoc/Ender3V2S1 #1014: [BUG] Missing proui library for `*_maple` environments - undefined references

classicrocker883, 2023-08-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

No errors and compiles successful using `STM32F103RE_creality` **default_envs** in *platformio.ini*  
except when trying to use `BOARD_VOXELAB_AQUILA` or `BOARD_CREALITY_V4` with `GD32F103RC_voxelab_maple` or `STM32F103RC_creality_maple` causes several errors.

```r
Marlin\src\lcd\e3v2\proui\toolbar.cpp: In function 'void drawToolBar(bool)':
Marlin\src\lcd\e3v2\proui\toolbar.cpp:56:16: warning: iteration 5 invokes undefined behavior [-Waggressive-loop-optimizations]
       TBGetItem(PRO_data.TBopt[i]);
       ~~~~~~~~~^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\e3v2\proui\toolbar.cpp:55:27: note: within this loop
     for (uint8_t i = 0; i <= TBMaxOpt; ++i) {
```
---
```r
D:\Users\Miguel\Documents\Trabajos\VSCode\Marlin_Ender3V2S1-Development/Marlin\src\lcd\e3v2\proui\../../../feature/runout.h:219: warning: undefined reference to `analogInputPin'
D:\Users\Miguel\Documents\Trabajos\VSCode\Marlin_Ender3V2S1-Development/Marlin\src\lcd\e3v2\proui\../../../feature/runout.h:219: warning: undefined reference to `digitalPin' 
D:\Users\Miguel\Documents\Trabajos\VSCode\Marlin_Ender3V2S1-Development/Marlin\src\lcd\e3v2\proui\../../../feature/runout.h:219: warning: undefined reference to `FastIOPortMap'
Marlin/lib/proui\libproui.a(proui.cpp.o): In function `ProUIEx::setRunoutState(unsigned long)':
D:\Users\Miguel\Documents\Trabajos\VSCode\Marlin_Ender3V2S1-Development/Marlin\src\lcd\e3v2\proui/proui.cpp:179: warning: undefined reference to `pinMode'
```
this is not even my project directory
>D:\Users\Miguel\Documents\Trabajos\VSCode\Marlin_Ender3V2S1-Development

GD32 architecture library is missing from **Marlin/lib/proui/**

>    gd32f10 = _GetMarlinEnv(MarlinEnv, 'MAPLE_STM32F1')
...
      elif (gd32f10):
         arch = 'gd32f10/'
         print ('GD32F1 Architecture detected')

### Steps to Reproduce

1. for **MOTHERBOARD** in *Configuration.h* - use `BOARD_VOXELAB_AQUILA` or `BOARD_CREALITY_V4` 2. for **default_envs** in *platformio.ini* - use `GD32F103RC_voxelab_maple` or `STM32F103RC_creality_maple`.
3. Build / compile firmware 

### Version of Professional Firmware

August 2023 (newest)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-08-07:
Please read the release notes of latest version:
- Maple support was deprecated

classicrocker883, 2023-08-08:
> Please read the release notes of latest version:
> 
> * Maple support was deprecated

thats unfortunate because the Aquila GD32 and N32 will not work without Maple environment

closed 2023-09-11 by mriscoc