mriscoc/Ender3V2S1 #1024: [BUG] Incorrect behavior when using z-probe

ksa-real, 2023-08-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have a generic 3d Touch sensor. First, by pointing to the same point at the bed I found and set in the "Probe settings" the X and Y probe offsets: -41.5 and -10.8. The difference of -3.8mm compared to the default -7mm is due to using a printed bracket: https://www.thingiverse.com/thing:4462870/files (V3_probe_mount.stl).

Then I went to Prepare->Z Probe Wizard (Probe Z offset was 0.00) and did Auto-Home. The final position was z=5.00. I deployed the probe (M401) and moved to z=0.00. The probe was well above the bed. This was already unexpected. Isn't Z-offset equal to position of the probe trigger point? As such moving to z=0 with probe z offset=0 should make the probe touch the bed.

Then I removed the software endstops (M211) and started slowly moving down in 0.1mm steps. The probe touched the bed at -3.6mm and triggered at z=-4.98 (I switched to 0.01 steps after -4.9). So z is incorrect and is 5mm less than it is supposed to be.

### Bug Timeline

I just installed 3d touch and new firmware (Ender3V2-422-BLTUBL-MPC-20230809.bin instead of Ender3V2-422-MM-MPC-20230805.bin)

### Expected behavior

I expected the probe to touch the bed at position z=0.

### Actual behavior

It touched the bed at position approximately z=-5.0

### Steps to Reproduce

See description

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-422-BLTUBL-MPC, based on bugfix-2.1.x (Aug  9 2023 21:48:38)

### Printer model

3V2

### Electronics

4.2.2

### Add-ons

3d touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-08-14:
With a Z probe offset of 0 (`M851 Z0`), Z at 0 (`G0 Z0`) will move the Z axis to the probe trigger point.

ksa-real, 2023-08-14:
It doesn't. It moves 5mm above the triggering point.

mriscoc, 2023-08-14:
Tested and it works as intended in current release [Ender3V2-422-BLTUBL-MPC-20230809.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230805/Ender3V2-422-BLTUBL-MPC-20230809.bin)

ksa-real, 2023-08-14:
What I just tried:
- Turn on the printer (no sd card or USB cable inserted)
- Control->Restore Defaults
- Reboot printer
- Prepare->Z-probe Wizard -> Auto Home
Printer now is actually in the correct position!

Now:
- Prepare->Z-probe Wizard ->Move Z to Home
Printer now 5mm above expected position. I.e. it reports "z=0" but is 5mm above the trigger point. Is that expected? May be I just don't understand this option.

ksa-real, 2023-08-14:
> > * Control->Restore Defaults
> > * Reboot printer
> 
> After restore defaults, save settings.
> 
> Move Z to home will move Z to the trigger point + Probe Z offset.

I've done that, forgot to mention in the steps. Fixed.

ksa-real, 2023-08-14:
Interestingly, that time when it worked it did probing twice: the first time it was pretty quick and the second time it was slower. Now, I see the printer doing probing only once. In such situations, there is an extra 5mm offset.
Now, even if I try to reset and save settings, I still see the same behavior with an extra 5mm offset.

mriscoc, 2023-08-14:
Sorry but I can't reproduce your issue here. Maybe there is a problem/compatibility issue with your probe.

mriscoc, 2023-08-14:
The blinking indicates some problem with the probe, sometimes is a white wire loose connection.

ksa-real, 2023-08-18:
Closing this for now. There is certainly some problem with my probe (likely hardware).

ksa-real, 2023-08-18:
Closing this for now. There is certainly some problem with my probe, likely hardware.

closed 2023-08-18 by ksa-real

ksa-real, 2023-10-11:
Just in case someone has the same issue: in my case the probe doesn't seem to support high-speed probing mode. I had to uncheck the "High-Speed mode" which is ON by default.