mriscoc/Ender3V2S1 #1025: [BUG] (Keeps going to Runout Senor purging screen)

hagenman, 2023-08-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

After upgrading, prints will run for a bit, then the runout sensor will trigger (i think?) and it will go to the purging screen. I tell it to continue then it will for a little while before the issue occurs again.  Also, the estimate time just keeps climbing until the sensor triggers again.

### Expected behavior

I expected prints to continue like they did on your previous firmware

### Actual behavior

After upgrading, prints will run for a bit, then the runout sensor will trigger (i think?) and it will go to the purging screen. I tell it to continue then it will for a little while before the issue occurs again.  Also, the estimate time just keeps climbing until the sensor triggers again.

### Version of Professional Firmware

20230810-11:42

### Printer model

Ender 3 S1 F4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

anoblet, 2023-08-14:
I'm having a similar issue. My prints will hit the runout screen, although there's still plenty of filament running through the sensor. Afterwards the filament breaks since it stays in the hotend heating.

classicrocker883, 2023-08-14:
check the runout setting, if set to HIGH/LOW set to opposite

hagenman, 2023-08-14:
Mine is set to Motion.

lam522008, 2023-08-15:
I am also having this issue on my Ender 3 v2, not sure what to do unable to print anything without it purging and spitting out the filament for no apparent reason. Any solutions?

mriscoc, 2023-08-15:
Check your filament runout sensor settings: https://github.com/mriscoc/Ender3V2S1/wiki/Filament-Settings-Menu#run-out-sensor

lam522008, 2023-08-15:
Thank you that did help! I don't have errors but can't seem to print anything more than 20 minutes without it messing up. Any solutions?

monkiki, 2023-08-15:
I've the same issue in my Ender 3 S1. Going back to v20230805 solved the problem.

lam522008, 2023-08-15:
> I've the same issue in my Ender 3 S1. Going back to v20230522 solved the problem.

How did you go back to that Version?

Ollipop030, 2023-08-15:
> How did you go back to that Version?

Here:
https://github.com/mriscoc/Ender3V2S1/releases/tag/20230522

Ollipop030, 2023-08-16:
> If you have the Ender 3 S1 and the stock filament runout sensor, go to Advanced / Filament settings menu and set Runout Active to HIGH

Doesn´t work for me. I´ve testet high/low, the printer stops every few minutes. Sometimes it doesn´t even finish the first layer. Only option for me is to disable the sensor. With May firmware it works fine.

NetCaptive, 2023-08-19:
Setting "Enable Runout" to "off" solved this problem for me on my Ender 3v2 with BLTouch.  My "Runout Active" setting was "Motion" and I left that unchanged.

andrenalin282, 2023-08-21:
Same problem for me Ender 3V2, high, low, runout deactivation, nothing works, plugged off the cable works for me, i wait for the fix.

mriscoc, 2023-08-21:
> Same problem for me Ender 3V2, high, low, runout deactivation, nothing works, plugged off the cable works for me, i wait for the fix.

The problem was only with the default value of Runout active, deactivating or set it properly should solve the issue.

RejZoRSheep, 2023-08-24:
Jesus, wasted almost 1 hour trying to figure out the problem, then finally come here and see it's not just me. The buggy update with invisible toolbar didn't cause me any problems, updated to latest fixed version and haven't printed anything then. Tried now and kept pausing even though I just put new 1KG spool on it.

plastisizer, 2023-08-25:
Had the same issue with the Advaced pause, turned off the runout detector and now was able to move past that point. I was then able to start the print, but he extruder was not pushing any filament. I rebooted and was able to restart the print without issues.

Crapoulos, 2023-08-29:
If it helps, I also had the same problem on my Ender 3 S1 with the latest version 20230805.
When I re-installed version v20230522 it solved the problem. 
I tested version 20230805 again and the problem returned. Runout sensor comes back every 4mn or so. Not yet tested the suggestion "set your Runout Active to HIGH or LOW".

Rippeee, 2023-08-30:
Has anyone found a solution for BTT yet? Other than rolling back previous version?

Belliqueu, 2023-08-30:
Disabling runout in the menus did not work for me, but adding M412 S0 to my start code did

mriscoc, 2023-08-30:
After disable it in the menu you need to save settings.

Rippeee, 2023-08-31:
If I Disable it from the menu, then it's disabled, If I enable it, it triggers pause like every 30 seconds. How do I get Smart sensor to work like intended?

Rippeee, 2023-08-31:
For those who are having issues with BTT sensor triggering every 30sec, I think I found the solution. I'm not using any custom configs or G-godes. I'm using the latest mriscoc firmware with BTT sensor, Mode set as motion. The recommended Distance for BTT is 7 mm, and that doesn't work, unless your sensor is literally Next to the extruder. I would recommend to set it to 14 mm, if that doesn't work, try 20,30,40,50 etc. In my case, I'm running Sprite pro Direct drive upgrade kit, and I'm mounting the sensor at the top, next to the filament spool. For that reason, there is a lot of ''slag'' between the extruder and the sensor, so there will be moments when the filament ain't moving at all. 7 mm would never work in my case and in my case it doesn't really matter even if the distance is 100 mm because once the filament is out of sensor, there is still like 20CM of filament going in the extruder. I set my distance to In a Nutshell the solution is Crank up your distance from 7 to 8,9,10,11,12,13,14,15,16… etc.   Until it works.

mriscoc, 2023-09-06:
Solved in the last version

closed 2023-09-06 by mriscoc

Crapoulos, 2023-09-09:
Thank you Miguel for all your hard work and all the improvements you've made with all these firmwares ! Much appreciated.

Rippeee, 2023-09-09:
Little of the topic but if some of you're running BTT Smart filament sensor with a Sprite extruder and Looking for a mounting system that allows to have that 7 mm Run out distance, and actually WORK

You are welcome…  https://www.thingiverse.com/thing:6209226

zfkdnop, 2023-10-23:
> Solved in the last version

Now I can't compile:

`Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_sram.cpp.o
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_wired.cpp.o
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/fast_pwm.cpp.o
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/fastio.cpp.o
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/msc_sd.cpp.o
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/sdio.cpp.o
Compiling .pio/build/STM32F103RC_creality/src/src/HAL/STM32/timers.cpp.o
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/HAL_SPI.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/fastio.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/Servo.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_bl24cxx.cpp:32:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_sdcard.cpp:31:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_if_iic.cpp:32:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_flash.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_sram.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/eeprom_wired.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/fast_pwm.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
[... 10 lines cut ...]
                 from Marlin/src/HAL/STM32/timers.cpp:26:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2753:6: error: #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
 2753 |     #error "RUNOUT_TUNE_ITEM requires HAS_FILAMENT_SENSOR."
      |      ^~~~~
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/fastio.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_sdcard.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_if_iic.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_wired.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_sram.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_creality/src/src/HAL/STM32/eeprom_flash.cpp.o] Error 1
[... cut ...]
=========================================================================== [FAILED] Took 8.66 seconds ===========================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_creality  FAILED    00:00:08.657
===================================================================== 1 failed, 0 succeeded in 00:00:08.657 =====================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`