mriscoc/Ender3V2S1 #1031: [BUG] Nozzle Wiping off bed

oliver408i, 2023-08-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

On the Ender V2 Neo, the firmware works fine up until a print is started and it starts a nozzle wipe. The offset for the wipe is too far to the left and goes off the buildplate.
Although this is easy to fix in Cura by increasing the start g-code's X value, it make me wonder why it happens on this firmware and not the original firmware. It could be that the entire grid if offsetted toward the left, which could affect big prints.
As of now, all the prints I did came out fine.

### Bug Timeline

Aug 17, 2023

### Expected behavior

I expected it to be more right and stay on the buildplate, just like how the original firmware does it.

### Actual behavior

It moved too far to the left and extruded filament off the buildplate

### Steps to Reproduce

1. Obtain a Ender-3 V2 Neo
2. Flash the firmware onto it (via sd card by just putting the .bin into the root dir)
3. Use the Cura Ender-3 / Ender-3 v2 profile to slice any print
4. Upload it via SD card and start a print
5. Observe the wipe going off the side

### Version of Professional Firmware

20230805

### Printer model

Ender-3 V2 Neo

### Electronics

Board 4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

Default Cura Profile start g-code

```
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
```

Suggested modified lines:
```
G1 X5.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X5.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X5.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X5.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
```

mriscoc, 2023-08-17:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

closed 2023-08-17 by mriscoc

oliver408i, 2023-08-17:
It doesn't seem like my print has an advanced > physical settings option inside of control nor prepare.