mriscoc/Ender3V2S1 #1033: [BUG] Mesh & Tramming Region Stuck at Smallest Previous Bed Size

EthanMWoodruff, 2023-08-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When adjusting the X and Y bed sizes in the advanced settings tab the smallest previously selected bed size appears to "stick". For example if I set the bed size to 150x150mm then adjust it back to 230x230mm the 150x150mm bed will be used in any following tramming or bed mesh commands. This includes the built in tramming wizard, auto build mesh function, or by manually running the `G29 P1` command in Octoprint.

This "stuck" state can be fixed using the restore defaults button in the control tab. Physically powering on an off the printer, as well as using the reboot printer button in the control tab only fix the issue if the settings haven't been stored while the bed is in the "stuck" state.

### Bug Timeline

This bug exists as of at latest version 20230805

### Expected behavior

I expected the tramming wizard to cover the entire print bed.

### Actual behavior

The tramming wizard only covered the front left corner of the print bed.

### Steps to Reproduce

1. Decrease the bed size in the advanced settings tab.
2. Increase the bed size in the advanced settings tab.
3. Run the tramming wizard or auto build mesh function.

### Version of Professional Firmware

20230805

### Printer model

Creality Ender 3V2

### Electronics

4.2.2

### Add-ons

BL Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

mriscoc, 2023-08-18:
Maybe you forgot to adjust the mesh inset:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

EthanMWoodruff, 2023-08-18:
> Maybe you forgot to adjust the mesh inset: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

Yep that seems to be the issue. Sorry about the false report and thank you for the quick response!

closed 2023-08-18 by EthanMWoodruff