mriscoc/Ender3V2S1 #1034: Z-offset is inconsistent (for me) when using Manual Mesh

Ergonomicmike, 2023-08-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I just updated to the August BugFix 1 update of Professional firmware. (But this issue has happened before.)

Perhaps I am doing something wrong when I use Manual Mesh to probe the bed. But the value of Z-offset for the mesh changes about 0.28 mm between the first Mesh Leveling to Editing the Mesh. And between powering off and powering on the printer.

I made a video to show the problem and when it happens to me.

https://jmp.sh/p5Yl5tMM

(If too long, jump to 5:54 in the video.)

### Bug Timeline

I have noticed an inconsistency in Z-offset for Manual Mesh since I first tarted using the Professional Firmware. (March 2023)

### Expected behavior

I expect that there will be only one Z-offset for the bed/mesh.

### Actual behavior

As in the video, there is one value of Z-offset needed during Manual Mesh. But then another value is needed when editing the exact same Mesh.

And then when the power is turned off to the printer and turned back on, the first Z-offset value is needed. Until another power down/reset, when the second value is needed.

### Steps to Reproduce

Please see the video at
https://jmp.sh/p5Yl5tMM

### Version of Professional Firmware

Version 2.1.3.MRiscoC

### Printer model

Ender 3V2

### Electronics

4.2.2

### Add-ons

None

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-08-18:
Seems to be an inconsistent Z end-stop switch.

Ergonomicmike, 2023-08-18:
Thanks for the quick reply.

I thought of that. And I will check it again now. But, IIRC, the Bed Tramming always is the same - 0.020 mm using my metal gauge. If the stop switch were inconsistent, then the Bed Tramming should be changing all the time too.

But I'll check it again a few times now.

mriscoc, 2023-08-18:
Also, you don't need to power off the printer, there is a reboot command in the Control menu, or if you have a terminal, send M997.
If the Z end-stop switch is inconsistent, its faults will be evident between Z homes.

Ergonomicmike, 2023-08-18:
Okay, I used an air duster to blow out/clean out the switch. Then I sprayed the switch with an electrical lubricant used in Aviation electronics. Mechanically, the switch is secure.

I performed about 20 tests, some using the Bed Tramming, and some using Home Z and Disable. All give repeatable results using the metal feeler gauge.  (If my end-stop switch were inconsistent, I expect that the inconsistency would show after the "Home Z and Disable" up and down movement.)

So now I'm wondering if this is an issue in Marlin? You would know the demographics of your downloads. But I presume that most users of mriscoc firmware have ABL. So perhaps this is an under-reported issue? Perhaps the averaged Z-Offset changes with the last point that was manually probed? 

Perhaps I should post this in the Marlin github?

The reason that I did the power down/power up was simply to show that the problem occurs when the firmware restarts, after any "bad" data is presumably flushed/cleared.

If it were a problem with my micro-switch, it is strange that it is always by the same amount. I expect that an inconsistency in a switch would be somewhat random over a range. But this is either 0.070 or 0.043 in my unit.

(BTW, I said in my video that my metal feeler gauge was 0.020 mm. It is not. It is 0.020 inches.)

Ergonomicmike, 2023-08-18:
Testing the repeatability of the Z-stop switch

https://jmp.sh/UeGttAn0

mriscoc, 2023-08-18:
From your video, the repeatability of the Z end-stop switch looks good.  It might be a good idea to load the latest Marlin bugfix and try the same tests.

Ergonomicmike, 2023-08-18:
Okay, I will try it.

(Also, if the problem were my Z stop switch, then the probe point measurements wouldn't have been repeatable. But they were.)

Ergonomicmike, 2023-08-19:
I'm going to show my ignorance.

I compiled the latest Marlin and I hacked the configuration.h for Manual Mesh, using your configuration.h as a guide.

The firmware built okay. But there's no way (that I can find) to set the Manual Mesh from the UI.

I'm guessing that Marlin expects users to use Octoprint for Manual Mesh?

Or maybe the Marlin UI? (I'm using the Creality UI.)

Either way I don't have either. So I might be at an impasse.

mriscoc, 2023-08-19:
You can use the MM configuration files from: https://github.com/mriscoc/Special_Configurations in to the Marlin sources.

Ergonomicmike, 2023-08-19:
Thanks. Working on it now.

Ergonomicmike, 2023-08-19:
Okay, I have the latest Marlin Bugfix working. One thing that's different is that the Z-Offset isn't "Live" in the Marlin Bigfix version. That is, changing the Z-Offset doesn't move Z in real time.

But así son las dos de la mañana acá, I might find, after some sleep, that I did something wrong in the build.

Ergonomicmike, 2023-08-19:
I am making some progress. I found how to enable live baby-stepping. I am now obtaining the same Z-offset (-0.7) as I was in the first step of my video when I was using MRiscoC Firmware. 

And although I have enabled the Edit Mesh Menu, I cannot make Marlin Bugfix move the print head to the point I want to edit. 

Is that a feature unique to MRiscoC firmware?

As it is now, if I wanted to edit a point on the Mesh, I have to Re-level the entire Mesh.

But I'll continue to experiment.

mriscoc, 2023-08-19:
> But así son las dos de la mañana acá,

Si hablas en español podemos continuar en español si lo deseas.

It right, some features are unique to Professional Firmware (PF) and with the time some of them are ported to the Marlin project, but the PF will have always the latest enhancements.

Ergonomicmike, 2023-08-19:
¡Success! (Using MRiscoC firmware, August Bugfix 1.)

I "hacked" at both .h files and something that I changed (or more than one thing that I changed) solved the problem. I now have consistent Z-Offset after editing the (Manual) Mesh.

I will plan to attach the current and previous .h files later (for WinMerge purposes) and then will close this issue.

Ergonomicmike, 2023-08-20:
Attached are the Before and After files (Before I had the Z-Offset inconsistency, and now After, now that it's fixed) of configuration.h and configuration_adv.h.

I'll save you time by saying that the only two changes that appear related to this issue are:

I disabled SHOW_REAL_POS in configuration.h

And I enabled OPTIMIZED_MESH_STORAGE in configuration_adv.h

(I'm using a 4x4 Mesh in a 256K Environment. But I was having the inconsistent Z Offset with a 3x3 Mesh too.)

Oh, and for completeness, Visual Studio Code updated some parts of itself during the time that I was trying to fix this issue.

(As an aside, there were two settings (JD_TUNE_ITEM  and ADVK_TUNE_ITEM) that I should have had enabled Before, but hadn't. Interestingly, those menu items appeared on my Display anyway and uncommenting the lines didn't appear to change anything on my Display. I'll mention this in a Discussion topic later.)

-----------

Now, I should undo these two above changes, one at a time, to see which one causes the Z-offset issue to return. And I might do that later.

But for now, since I solved the issue that I was having and am able to print consistently now, I leave the finding of the cause as an exercise for the reader.

[Configuration.h_Before_and_After.zip](https://github.com/mriscoc/Ender3V2S1/files/12387289/Configuration.h_Before_and_After.zip)
[Configuration_adv.h_Before_and_After.zip](https://github.com/mriscoc/Ender3V2S1/files/12387290/Configuration_adv.h_Before_and_After.zip)

Ergonomicmike, 2023-08-21:
I'm so embarrassed. (But not 'embarazada.')

After I updated the firmware, I made about ten prints, all with consistent Z-offset and perfect first layers.

After I finished my print projects, I decided to try to find out what change in the configuration .h files made the improvement since the time that I recorded my video.

So I un-did the changes that I had made to the configuration files, one by one, trying each revised firmware. 

No change.

Stymied, I reverted to the firmware that I was using in the video I had made for this issue.

Suddenly the Z-offset changed! From -0.64 to -0.09 !

But unlike the video, the Z-offset was consistent.

I don't understand how merely changing firmware can cause Z-offset to change. I didn't touch anything mechanical.

I can't believe that there was previously some kind of flaw in Visual Studio Code that caused a compile error that affected calculated Z-offset. (However, all the "undo's" above where compiled with the latest, updated VSC.)

So perhaps el Señor Miguel is correct and it's my Z limit switch?

(Or maybe it's heat from the bed affecting the switch? Because when I performed the tests above, I decided to save time by not waiting for the bed to heat. I measured Z-offset on a cold bed.)

I visually inspected the switch PCB closely, hoping to find a bad solder joint. But it looks good to me. (Although my old eyes cannot see if the surface mount resistors are okay.)

I lowered the speed of the Z homing, hoping that this will minimize any point bounce from the switch - if there is any.

For now, this seems like a búsqueda inútil. (A wild goose chase.) Closing the (non- ?) issue now.

closed 2023-08-21 by Ergonomicmike