mriscoc/Ender3V2S1 #1041: [BUG] Ender 3 S1 Plus can't get CVLM to work

aerladis, 2023-08-20:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Ok so I got the released CVLM version for my F4. There are 2 alternative I want to use my laser.
However once I plug my laser Z axis doesn't move downwards to focus. 
When using Lightburn I auto home I keep getting error:
Error:!! STOP called because of BLTouch error - restart with M999
When I use the SD card to transfer Gcode files I can home X-Y axes run range but I cant engrave because I cannot focus Z. 
It seems  like The machine needs to be in Laser mode similar to creality version So that I can move Z axis freely and bl touch check is disabled.

### Bug Timeline

Current bug

### Expected behavior

Printer is in laser mode Where I can move Z axis to - to focus. 

### Actual behavior

Z axis doesn't move down due to firmware reporting bl touch is missing.

### Steps to Reproduce

1. Flash CVLM Firmware
2. Plug in the laser module
3. Try to focus laser with moving z axis down.
4. Z axis doesn't move down and prompts error :Error:!! STOP called because of BLTouch error - restart with M999

### Version of Professional Firmware

Ender3S1-F4-CVLM-20230812

### Printer model

Creality S1 PLus

### Electronics

4.2.2 F4

### Add-ons

CVLM 5w Laser module

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

mriscoc, 2023-08-21:
For focus adjust, use **Laser Focus** item in the Laser Control menu:

[image]

aerladis, 2023-08-21:
> To adjust the focus, use **Laser Focus** item in the Laser Control menu:
> 
> [image]

It doesnt't go down only goes up.

mriscoc, 2023-08-21:
You must to do an autohome before of put the laser module, or disable motors, move the z axis manually and use Set as Home (At the bottom of the menu)

aerladis, 2023-08-21:
Oh i see. Is lightburn supported?

mriscoc, 2023-08-21:
> Is lightburn supported?

No one has tested CVLM support with Lightburn. But always there is a first time, maybe it is necessary to do some adjustments. I think that the Creality Ender 3 S1 CVLM firmware also lacks support for Lightburn.

aerladis, 2023-08-21:
> > Is lightburn supported?
> 
> No one has tested CVLM support with Lightburn. But always there is a first time, maybe it is necessary to do some adjustments. I think that the Creality Ender 3 S1 CVLM firmware also lacks support for Lightburn.

I tested now even when I auto-home The Z axis lifts and takes that point az 0. Because I cannot go - I cannot focus the laser.

mriscoc, 2023-08-21:
I don't have a valid license for Lightburn so I'm afraid I can't help you. But the last time that I check, Lightburn was creating a totally different code than the Creality Laser software.

aerladis, 2023-08-21:
> I don't have a valid license for Lightburn so I'm afraid I can't help you. But the last time that I check, Lightburn was creating a totally different code than the [Creality Print / Laser software](https://www.creality.com/pages/download-ender-3-s1-pro?spm=..page_1934481.products_display_1.1&spm_prev=..page_1967279.header_1.1).

Sir, I don't want this thread to get out of its purpose but;
1: I have been working on ABL issues on S1. In my case it was simply oily bed a wash with the dishwasher solved the issue after I tighteneed/loosened every screw on the machine for the past week. I am sure although as simple as it looks many people skip this and overcomplicate the issue. So please note this on Calibration Guide.
2. Lightburn doesn't work at all and spits out the BLtouch error I mentioned in the first post. It offers a free trial for a month  if you want to test things.
3. After homing Z raises and I cant dial it back when CVLM is plugged. The firmware look for CRtouch as a Z stop sensor I think. This is independent from Lightburn.
Thanks for your responsesç

mriscoc, 2023-08-21:
CVLM support thread: https://github.com/mriscoc/Ender3V2S1/discussions/62

aerladis, 2023-08-21:
> CVLM support thread: #62
> 
> 3.- Yes, the firmware (Ender3S1-F4-CVLM) expects to have the CRtouch installed when not in the laser mode.

Oh, Is there a Laser Mode? I only noticed the Laser Control Menu.

kpbicka, 2023-08-28:
> ### 
> Ok so I got the released CVLM version for my F4.  
> ### Version of Professional Firmware
> Ender3S1-F4-CVLM-20230812
> ### Electronics
> 4.2.2 F4
> ### Add-ons
> CVLM 5w Laser module

Dear friend! I am also using LightBurn software and it does not detect my printer. Can you please tell me what settings are you using?
my settings are like this:

mriscoc latest firmware [Ender3S1-F4-CVLM-20230812.bin](https://github.com/mriscoc/Special_Configurations/releases/download/CVLM/Ender3S1-F4-CVLM-20230812.bin)
1. device Marlin (not detected automatically at the beginning)
2. device setting:
Enable DTR signal : NO
Laser Control Commands: M03/M05
Baud Rate: 115,200

With this settings official Creality Firmware Ender-3S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER_0506.bin works fine

mriscoc, 2023-08-28:
Try with baud rate at 250000

kpbicka, 2023-08-28:
> Try with baud rate at 250000

Wow! It solve my problem. Thank you a lot!

mriscoc, 2023-08-28:
In Advanced/Laser Control 
Use Toggle Laser for change from 3D printer to Laser engraver.

closed 2023-09-11 by mriscoc