mriscoc/Ender3V2S1 #1042: [Setting] MINTEMP error immediately after print start

csihama, 2023-08-20:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

On Ender 3 V2 with 4.2.2 board, can't heat up hotend and bed at the same time on lots of FW (sometimes works, sometimes not). Need to change the start G-code script: 
`M190 S{material_bed_temperature} ; heat to bed setting in Cura and WAIT
M109 S{material_print_temperature} ; heat hotend to setting in Cura and WAIT`

### Expected behavior

Heat both hotend and bed

### Actual behavior

MINTEMP error

### Version of Professional Firmware

Custom built from your repo

### Printer model

Creality Ender 3 V2

### Electronics

Board 4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

csihama, 2023-08-20:
Further investigating: 
`Recv:  T:29.91 /0.00 B:51.53 /0.00 @:0 B@:0
Recv:  T:29.91 /0.00 B:51.49 /0.00 @:0 B@:0
Recv:  T:29.95 /0.00 B:51.41 /0.00 @:0 B@:0
Recv:  T:29.96 /0.00 B:51.36 /0.00 @:0 B@:0
Recv: Error:MINTEMP triggered, system stopped! Heater_ID: 690
Recv: echo:Home XYZ First
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error: MINTEMP triggered, system stopped! Heater_ID: 690 - echo:Home XYZ First
Recv: echo:Err: MINTEMP
Recv: Error:Printer halted. kill() called!`

csihama, 2023-08-21:
Solved,

Need changes in Configuration_adv.h
`#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 1
#define PREHEAT_TIME_HOTEND_MS 30000`

mriscoc, 2023-08-22:
```cpp
#define PREHEAT_TIME_HOTEND_MS 30000
```
Maybe is not a good idea, check the recommendation in the configuration file.

classicrocker883, 2023-08-25:
> Solved,
> 
> Need changes in Configuration_adv.h `#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 1` `#define PREHEAT_TIME_HOTEND_MS 30000`

I leave these disabled.

csihama, 2023-08-25:
I played with the values. The lowest value was: 19000. If i go with lower values, the FW kills the machine with MINTEMP error, even while the temperature reading rising. 

Solved the issue for myself, the printer works with that timeout offset. Perhaps because my Ender 3 V2, board 4.2.2 with STM32F103RE was the first implementation. I don’t know, but thanks for the reply.

closed 2023-08-25 by csihama