mriscoc/Ender3V2S1 #1048: Bed doesn't cool down after print is finished

lkos98, 2023-08-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Firmware Ender3S1-F4-UBL-20230810. After print is finished bed temperature shows the set one and doesn't go to 0. 
Simple fix is to add "M140 S0" at the custom G-code in slicer, but it has to be done for all printer profiles. 
Also if bed temperature is set lower after the first layer/s, "bed cooling" shows on the screen and stays there until printing is finished. I did not test if I don't set lower temperature after the first layer - if this would fix the issue.  
With Ender3S1-F4-UBL-20221222 bed temperature was working fine.     

### Bug Timeline

After update to Ender3S1-F4-UBL-20230810

### Expected behavior

After print is finished bed temperature should go to 0.

### Actual behavior

Shows 57/57 (for example) while hot end goes properly to 0. 

### Version of Professional Firmware

Ender3S1-F4-UBL-20230810

### Printer model

Ender 3 S1

### Electronics

F4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Additional information & file uploads

I use mostly Superslicer

mriscoc, 2023-08-24:
The forced bed cooler at the end of the print was removed a few builds ago due to it was causing issues and being unable to be disabled by the user for some special processes. Now it is up to the user if he wants to cooldown and disable stepper drivers at the end of the printing by simply adding the proper g-codes in the end script as it is shown in the slicer scripts page of the wiki: https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts

mriscoc, 2023-08-24:
For the "bed cooldown" message, the latest F4 UBL version doesn't had enough free program memory to enable the status line timeout so the last message will be kept until a new message is sent, you could, for example, send the layer number using the M117 command from the slicer.

lkos98, 2023-08-24:
Ok, thanks for the fast reply. I will put M140 to all printer profiles. 
Firmware is amazing BTW, I tried the new Creality update (felt in the same trap - issue #966), and went back to  Ender 3 professional immediately - looks and works much better. 
Great work!