mriscoc/Ender3V2S1 #1052: [BUG] Serveral issues with Ender3S1-F1-UBL-MPC-20230809

Horgul123, 2023-08-26:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

When using the Ender3S1-F1-UBL-MPC-20230809 FW I have several problems:
The issue with the filament sensor is "fixed" with changing the setting to "LOW", but:
- sometimes I get the message that there is a heating problem and the print is paused. After reheating everything works again for a while
- Got messages like Nozzle moved to home without a reason and the print is paused

Going back to the 202305 FW fixes all problems, so it must be 202308 related.
Can't be reproduced because it happens not always and not always at the same time of of the print

### Bug Timeline

new

### Expected behavior

Work like previous version 202305

### Actual behavior

Strange errors:
Heating suddenly stopped
Nozzel moved to home and print is stopped

Going back to Ender3S1-F1-UBL-MPC-20230522B solves the problems

### Steps to Reproduce

starting  print

### Version of Professional Firmware

Ender3S1-F1-UBL-MPC-20230809

### Printer model

Ender3 S1 F1

### Electronics

?

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

jderosa3, 2023-09-01:
I can confirm that these issues exist. I flashed the Ender3S1-F1-UBL-MPC-20230809 FW as well and I get random actions on first layer where the head moves to back right corner of bed and reverses filament out. With no warning or reason why. I am testing a print on previous version.

Horgul123, 2023-09-01:
@jderosa3 Thanks for confirming! I was not sure if this is a problem on my side or a real bug.

mriscoc, 2023-09-01:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
[image]

mriscoc, 2023-09-01:
Duplicate of: https://github.com/mriscoc/Ender3V2S1/issues/1025

closed 2023-09-01 by mriscoc

jderosa3, 2023-09-01:
> https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides [image]

I have a run-out sensor and still got this error when active and sensor working fine.

Horgul123, 2023-09-01:
@mriscoc Sorry, I don't understand your hint. I saw the bug with the filament sensor and set it to "low". as @jderosa3 said, even with sensor and correct setting I got the error. There is no error message, the printer just behaves strange.
With the wrong setting for the sensor I get a filament error message. But now there is no filament related message, the printer just stoppes

mriscoc, 2023-09-01:
> I flashed the Ender3S1-F1-UBL-MPC-20230809 FW as well and I get random actions on first layer where the head moves to back right corner of bed and reverses filament out.

Read the Wiki (manual) and follow the guide about filament sensor.

mriscoc, 2023-09-01:
> I have a run-out sensor and still got this error when active and sensor working fine.

You need to found the correct High/Low configuration that match your filament sensor.

Horgul123, 2023-09-01:
@mriscoc : in my case I don't get a filament error when setting it to LOW. I get errors regarding a heating problem or without an error. This doesn't look like the filament sensor is the reason

mriscoc, 2023-09-01:
Heating is not related to the filament runout sensor.

mriscoc, 2023-09-01:
> No problem. Others have confirmed same issue on other forums as well.

@jderosa3 your heating issue is not related to the runout sensor, check your thermistor and heatblock silicone jacket.

I can't reproduce your heating issue.

Horgul123, 2023-09-02:
This is what we saying. But it only happens with your latest Firmware. With
the oder one everything works, somit can't be a Hardware issue

2023, 21:08:

mriscoc, 2023-09-03:
I need reproducible steps to be able to debug any possible bug. You are printing from SD or from a host (Octoprint)

Horgul123, 2023-09-03:
I'm printing via Octoprint.
It happens shortly after starting a print. I'll install the new version
again today and document everything in detail

Horgul123, 2023-09-03:
Checked it again as promised.
With "LOW" the print stopped after the first layer, purged some material and told me "Nozzle parked".
With "HIGH" it's working. Last time I had issues with this setting, too.

mriscoc, 2023-09-04:
> I'm printing via Octoprint. It happens shortly after starting a print. I'll install the new version again today and document everything in detail

Did you disable the +VUSB?

Horgul123, 2023-09-04:
@mriscoc no, didn't do this. It works fine.

At the moment it works with your latest FW, too. No idea, why it made problems a few days ago..... Did the same as yesterday.

Horgul123, 2023-09-04:
As mentioned: The printer runs fine. I use a network USB extention, so I don't think the pin is powered. I'll double check this