mriscoc/Ender3V2S1 #1055: AUGUST 2023

SergioFrank, 2023-08-28:
AUGUST 2023
=========

## What's new in this release ##
- LCD brightness dims after 5 minute timeout (not available in all versions)
- Slow retract item in Filament management menu
- Better bed tramming wizard
- Automatic homing axes before M48 (Probe test) if necessary
- Support to M86-M87 Hotend Idle Timeout
- Fixed C104 bug
- Fixed bug in UBL when an invalid point is found
- Improved UBL smart-fill
- M300 E1/0 enables or disables sounds
- C851 F edit probe Z fix compensation
- Fixed BLTouch deploy bug in HS mode
- Several minor bug fixes and Misc. optimizations
- New OFFICIAL/CUSTOM BUILD label in the info page
- and many more fixes from latest Marlin's bugfix 2.1.x

In Source:
- Added support to One cLick print https://github.com/MarlinFirmware/Marlin/pull/25781
- Preliminary support for E3 Free-runs board CR4NTxxC10 and other boards https://store.creality.com/products/e3-free-runs-tmc2209-32-bit-open-source-silent-motherboard
- Fixed Time Motion experimental support
- Improved support for Marlin Auto Build firmware xfer
- Maple support was deprecated

> **Note**1: In latest releases I'm promoting UBL versions over ABL, but ABL versions will be available through custom compile. If you had a `G29` to make an ABL leveling before each print change it to `G29 P1` for UBL. MPC is also being incorporated in replacement of the hot-end PID which will continue to be available on custom compiles.

> **Note**2: Due to lack of program memory space in the STM32**F4**01RCT6 SoC some features had to be disabled for the F4 UBL firmware version, for example: BAUD_RATE_GCODE, INDIVIDUAL_AXIS_HOMING_SUBMENU, SOUND_MENU_ITEM, LONG_FILENAME_WRITE_SUPPORT, BINARY_FILE_TRANSFER, POWER_LOSS_RECOVERY.

## CV LASER MODULE SUPPORT for Ender-3S1!!
https://github.com/mriscoc/Special_Configurations/releases/tag/CVLM
[image]

This firmware is based on [Marlin](https://github.com/MarlinFirmware/Marlin), has been optimized and has enhancements for the Creality Ender 3 V2/S1 printer. This is an **universal firmware** that can be used in both RET6 and RCT6 STM32 chips that have 256k or 512k Flash memory. The firmware and its wiki are not really intended for beginners but as a supplement to already existing guides.  

[How to install the firmware](https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware)  
   
[image]
[image]

## Requirements
In order to use this firmware, your printer must meet the following requirements:
- A compatible control board: Creality 4.2.2, 4.2.3, 4.2.7, V24S1_301 or SKR Mini E3 V3 for Ender series printers with a STM32 SoC.
- A DWIN, DACAI, TJC or SYNWIT color display with encoder knob. Touch or legacy monochrome screens are not supported.

*You can modify your printer to meet these requirements*. 

## Versions
- **Ender 3v2 MM**: if you have the Ender 3v2 near to stock then use this version (Manual Mesh).
- **Ender 3v2 BLT**: use this version if you installed a CR/3D/BLTouch level sensor in the Ender 3v2.
- **Ender 3S1**: use this version if you have an Ender 3S1 printer with CRTouch, (F1 is for the STM32F103 and F4 is for the STM32F401 SoC)

You can download the __latest binaries__ from the assets at the bottom of
the __[release page](https://github.com/mriscoc/Ender3v2/releases/latest)__.
To know if you have the last release go to the info page in the display of
your printer and compare the release build date with the name of the binary
file. The filename follows the convention:  
  
> Ender3V2-422-MM-**YYYYMMDD**.bin
  
Where YYYYMMDD indicates the date of the build. 422 refers to the mainboard
version **4.2.2**, 423 refers to the mainboard version **4.2.3**, and 427 is for the **4.2.7** mainboard version, if you got
a **4.3.1** board use the 422 version. SKRMINI version is currently for the
BTT SKR Mini E3 V3 board (needs a screen adapter).

For example, if you want a firmware for your Ender-3 V2 printer with a control
board version 4.2.2, CRTouch and support for UBL, you should download the file:

> Ender3V2-422-BLTUBL-YYYYMMDD.bin
[... 197 lines cut ...]
* MPC/PID auto tune progress on display with plot
* Leveling progress on display
* Boot screen for M997
* Transparent icons for better color theme adaptability
* New print g-code preview
* New print done screen

**A lot of additions and fixes from Bugfix development branch**

## More info on the wiki pages ##
- [**How to install the firmware**](https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware)

- [**Octoprint settings**](https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint)

- [**How to make the Manual Mesh**](https://github.com/mriscoc/Ender3V2S1/wiki/Manual-Mesh)

- [**How to install a 3D/BLTouch**](https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch)

- [**How to apply a color theme**](https://github.com/mriscoc/Ender3V2S1/wiki/Color-Themes)

https://user-images.githubusercontent.com/2745567/143180943-f69f5967-34c1-421f-86a1-466ecccab97d.mp4

<hr /><em>This discussion was created from the release <a href='https://github.com/mriscoc/Ender3V2S1/releases/tag/20230805'>Ender-3 V2/S1 Professional Firmware 20230805</a>.</em>

_Originally posted by @mriscoc in https://github.com/mriscoc/Ender3V2S1/discussions/1010_

classicrocker883, 2023-08-28:
this is a safety guard to raise the Z a few mm after homing. do you have the previous Z axis switch still installed? that should be unplugged. 
auto bed leveling isnt actually done in the air. it might start 10mm above, but takes the probe measurements as needed. if you want Z axis to be 0.00 after homing you would need to change that in the firmware source code. otherwise in Z offset wizard menu you can set the nozzle down to the center of the bed and that should do what you are looking to do.

SergioFrank, 2023-08-28:
Andrew thanks for your attention. There is nothing I do that brings the hotend closer to the bed. Each time I request Z homing, the hotend goes up 5mm, if I request manual leveling, the hotend positions itself in the air. Could you explain better how to change the firmware source code? Where should I change? I am a beginner and still have little knowledge in this area. Thanks if you can help me.

Enviado do Outlook<http://aka.ms/weboutlook>

plastisizer, 2023-08-29:
might it be that your table just needs to be brought up a bit as maybe the z axis can't go any lower due to physical restraints? ie use the thumbwheels to raise the table a bit so there is pleany of room for the z axis to make contact.

SergioFrank, 2023-08-29:
I don't think I'm able to explain my problem. Regardless of how the bed springs are adjusted, any command given to the Z axis makes it move 5mm upwards, so it always moves away, and no command makes it go down again. After going up every 5mm there is no way to go down the hotend except turning off the printer and turning the Z axes (I use two) manually. Does anyone have an idea what might be going on? Thanks.

plastisizer, 2023-08-29:
ok so you have a BL touch or similar device correct? (touch bed leveler I believe). so that comes down and touches the bed corrrect but you're seeing the nozzle 5mm above the bed correct?...So it sounds like you just need to dial in your z offset no?

SergioFrank, 2023-08-29:
Yes, it's a Trianglelab 3D Touch. I can put any value for the Z offset when commanding auto leveling the bed will go up 5mm and start leveling in the air, and when I finish doing that, if I command Homing all axes, it will go up another 5mm, if I request new leveling it will go up 5mm more and leveling will be 15mm high. Any idea?

Enviado do Outlook<http://aka.ms/weboutlook>

mriscoc, 2023-08-30:
Hello, check the probe connections (3D touch), the printer raises the Z axis to make room for deploy the probe's tip, then begins to lower the axis until a probe's trigger signal is received.

I think that the trigger signal is always there, so the printer doesn't lower the axis and only raises it.

SergioFrank, 2023-09-01:
I believe this is the problem, after your instructions, I tested it with a pin-by-pin multimeter from the motherboard to the 3D Touch connector and the white Z-Stop wire (OUT) is not conducting. I believe that because it is a very thin wire, it broke or came with a manufacturer's defect. I will provide another one and I will post the result here again and my thanks for the commitment and dedication of Miguel Risco and Github collaborators. Thank you very much this was already driving me crazy.

Enviado do Outlook<http://aka.ms/weboutlook>

SergioFrank, 2023-09-04:
Muito obrigado. Agora funciona bem. Não conseguia entender como um produto novo poderia vir com defeito em seus cabos, Grato a todos colaboradores em especial Miguel Risco pela excelente versão do firmware.

Enviado do Outlook<http://aka.ms/weboutlook>

closed 2023-09-05 by mriscoc