mriscoc/Ender3V2S1 #1063: [BUG] (bug summary) speed creasing and decreasing

yourmomlol830, 2023-09-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When I turn it on The speed goes to 100 then 50 and keeps going nonstop even when printing and sometimes 100 to 25 

### Bug Timeline

It started when I first install the firmware 

### Expected behavior

For the speed to stay at 100 at all times

### Actual behavior

I have no idea 

### Steps to Reproduce

I don’t know 

### Version of Professional Firmware

2.1.3

### Printer model

Ended 3 s1 f4 chip

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

yourmomlol830, 2023-09-01:
Need help

mriscoc, 2023-09-01:
https://github.com/mriscoc/Ender3V2S1/wiki#firmware-features

classicrocker883, 2023-09-01:
to be more specific:  
this is a feature.

the flashing back and forth is between Speed % (default 100), and Speed in mm/s. (set by slicer).

it you compile your own firmware you can disable the feature with disabling `SHOW_SPEED_IND`

jxer, 2023-09-19:
> to be more specific: this is a feature.
> 
> the flashing back and forth is between Speed % (default 100), and Speed in mm/s. (set by slicer).
> 
> it you compile your own firmware you can disable the feature with disabling `SHOW_SPEED_IND`

where was this listed in change logs?

mriscoc, 2023-09-20:
> where was this listed in change logs?

Yes.