mriscoc/Ender3V2S1 #1066: [BUG] M109 not working as intended

Skyllarkk, 2023-09-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Im currently attempting to use M109 R45 to let to hot end cool down to 45 degrees before moving onto a move step. It will set the temp for 45 degree but it wont wait and will continue to move into the position asked . 

the end code im using is : 
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+2, max_print_height)} F600 ; Move print head up{endif}
G1 X5 Y{print_bed_max[1]*0.85} F{travel_speed*60} ; present print
{if max_layer_z < max_print_height-10}G1 Z{z_offset+min(max_layer_z+70, max_print_height-10)} F600 ; Move print head further up{endif}
{if max_layer_z < max_print_height*0.6}G1 Z{max_print_height*0.6} F600 ; Move print head further up{endif}
M140 S0
M109 R45
M211 S0 ; disable move limits
G1 Z243 F2000 ; SHUTDOWN USING SWITCH
M211 S1 ; enable move limits if printer is On
M107 ; turn off fan
M84 X Y E ; disable motors

any help would be great ,

cheers 

### Expected behavior

doesnt do anything until temperature of the hotend reaches 45 degrees

### Actual behavior

Moves onto the next step without waiting

### Steps to Reproduce

use above g code

### Version of Professional Firmware

2.1.3

### Printer model

ender 3 v2 neo

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Additional information & file uploads

ive tried with multiple slicers :)

mriscoc, 2023-09-03:
I tried this little script and M109 is working as intended:
```cpp
G28
M109 S50
M109 R45
G0 X100
```

Skyllarkk, 2023-09-03:
any reason you can think of that would cause it not to work in my end code
then ?

mriscoc, 2023-09-03:
Try with the little script and if it works start to add line to line until your get an error.

Skyllarkk, 2023-09-03:
i tried adding this to my script , still just continued to move without waiting to cooldown

mriscoc, 2023-09-03:
Use a gcode file with ONLY the lines I show in the little script.

Skyllarkk, 2023-09-03:
just tried it , it seems to work fine from what i can see , what is different about having that in my end code ?

closed 2023-09-11 by mriscoc