mriscoc/Ender3V2S1 #1067: [BUG] Unload filament option makes a brief grinding noise.

aerladis, 2023-09-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have an S1 F4 with latest firmware. When I unload filament extruder makes a grinding noise for a second, pauses and unloads filament succesfully. I suspects there is something incompatible with the first part of macro as after the grinding noise gears work perfectly in both ways an eject the Filament. The issue is present with no filament loaded as well. Other printing quality is unnaffected .

### Bug Timeline

New

### Expected behavior

Extruder silently ejects the filament. 

### Actual behavior

Extruder makes a grinding noise at the beginning of unload process then ejects it normally. After initial grinding gears work perfectly both way.

### Steps to Reproduce

Preheat hotend
Unload Filament
Grinds for a second
Extrudes and retracts normally until Filament is released.

### Version of Professional Firmware

Ender3S1-F4-20230809

### Printer model

Ender3 S1 Plus 

### Electronics

4.2.2 F4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

aerladis, 2023-09-02:
I tried Lowering E Max Speed and grinding noise decreases it goes away at 40mm/s but it is strange that I use 50mm/s retraction speed and never encountred grind noise. I suspect unload macro tries to spin the gears faster than the max speed at that inital 1 second for a reason.

plastisizer, 2023-09-08:
I hear the same issue, I notice that the extruder motor initially purges a small amount, which is when I hear the noise the most, and then a quick retraction untill the filament is removed. sounds like its just the rollers moving the filament moving super fast...

mriscoc, 2023-09-08:
It is not a firmware bug; you need to configure the extruder maximum acceleration and speed according with your hardware.

aerladis, 2023-09-08:
> It is not a firmware bug; you need to configure the extruder maximum acceleration and speed according with your hardware.

Do you know what settings Should I use for S1 pro?Couldnt find them anywhere.

classicrocker883, 2023-10-18:
> Do you know what settings Should I use for S1 pro?Couldnt find them anywhere.

in `ADVANCED_PAUSE_FEATURE`  
have you tried changing any of these?
```y
FILAMENT_CHANGE_UNLOAD_***
FILAMENT_CHANGE_LOAD_***
FILAMENT_CHANGE_FAST_***
FILAMENT_CHANGE_SLOW_***
```
whatever is pertaining to your specific issue with `***_LENGTH`, try lowering that value.  
because even for
```y
#define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                // This short retract is done immediately,
```

so correct me if im wrong but when the extruder gear first retracts, it does so fast, and it is this speed/distance that cannot keep up with the filament - so it grinds.

you can decrease the speed or distance and see what works.

closed 2023-12-29 by github-actions[bot]