mriscoc/Ender3V2S1 #1071: [BUG] Endstops not triggering when SD card is inserted.

colgateam, 2023-09-03:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

I'm using the creality free-runs board (STM32F401RE) on an Ender 3v2 with sprite pro extruder and CRTouch.

When an SD card is inserted, the x axis endstop will not trigger when pressed manually or while homing (cause the carriage to crash until power is removed), however all other axis' will work fine.
When I remove the SD card, the endstops all work as normal and the printer will home.

Has anyone encountered this before?
I'm not sure if it is a firmware issue or not though as there is no pre-compiled firmware for this board (official or unofficial).

### Expected behavior

I expect the endstops to all show TRIGGERED when activated.

### Actual behavior

The endstop stayed as OPEN when activated

### Steps to Reproduce

1. Insert SD card.
2. Home printer.
3. Watch crash.

### Version of Professional Firmware

August 2023 Bugfix 1

### Printer model

Creality Ender 3V2

### Electronics

Creality free-runs STM32F401RE

### Add-ons

CRTouch, Sprite Pro Extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-09-04:
The support for E3 Free-Runs is experimental, I will publish new sources soon.

colgateam, 2023-09-04:
> The support for E3 Free-Runs is experimental, I will publish new sources soon.

Thanks look forward to it, I assume you already know the other big issues with getting the board working like the marlin pin outs for BLTouch being the wrong way round etc.

mriscoc, 2023-09-04:
Where are you getting the config files from? You can use the configurator.pyw with the following json:
[CR4NTF4.zip](https://github.com/mriscoc/Ender3V2S1/files/12509138/CR4NTF4.zip)
Put the "CR4NTF4.json" file inside of _boards folder.

colgateam, 2023-09-04:
I basically took an ender3s1 config file and heavily modified it manually. It works but I’ll try one using the configurater tonight and see how it goes.

The biggest issue I had was the finding the following, in getting bltouch/crtouch working.

https://github.com/CrealityOfficial/E3-Free-runs-Silent-Motherboard/issues/4#issuecomment-1595010183

mriscoc, 2023-09-04:
This build is with my current configs
[Ender3V2-CR4NTF4-UBL-T13-20230903.zip](https://github.com/mriscoc/Ender3V2S1/files/12509287/Ender3V2-CR4NTF4-UBL-T13-20230903.zip)

colgateam, 2023-09-04:
I just tried the latest bin you sent, the sd card issue still persists and also the end stops are inverted so I wasn’t able to attempt homing.

[https://imgur.com/a/cjSNGVG](https://imgur.com/a/cjSNGVG)

mriscoc, 2023-09-04:
End-stops non inverted, ENDSTOP_INTERRUPTS_FEATURE disabled as in Creality example
[Ender3V2-CR4NTF4-UBL-T13-20230904B.zip](https://github.com/mriscoc/Ender3V2S1/files/12509824/Ender3V2-CR4NTF4-UBL-T13-20230904B.zip)

colgateam, 2023-09-04:
It thinks it has hit the end stop and homes in it’s current position.

[https://imgur.com/gallery/ZtJErQ9](https://imgur.com/gallery/ZtJErQ9)

mriscoc, 2023-09-04:
What about the previous issues?

mriscoc, 2023-09-04:
> It thinks it has hit the end stop and homes in it’s current position.
> 
> https://imgur.com/gallery/ZtJErQ9

In your last video, the dash board doesn't show 0 but - ? -, so, the homing operation was aborted.

colgateam, 2023-09-04:
> What about the previous issues?

They were still there.
I went back to my self-compiled firmware which only has the sd card error as shown.
[https://youtube.com/shorts/oFC0Ec8hAnA?si=v98tk9q_9ovK3UxN](https://youtube.com/shorts/oFC0Ec8hAnA?si=v98tk9q_9ovK3UxN)

The only way I can print is by using octoprint or pronterface.

mriscoc, 2023-09-04:
Can you post your configuration filesm

Ptitmile, 2023-11-13:
Still not fix homing with no problem but when you insert a sd card X endstop stop responding

marcus-e-e, 2023-12-19:
I have a very similar problem with this board on my stock Ender3 v2 (TJC display).

When I insert an SD card I get the media init failure and x,y endstops stop working.  
x endstop is none responsive. (if triggered, it changes to open)
y endstop still works
z endstop is none responsive (if open, it changes to triggered. 

[Ender3v2_config.zip](https://github.com/mriscoc/Ender3V2S1/files/13720083/Ender3v2_config.zip)

mriscoc, 2023-12-21:
Marlin recently incorporates support for that board, please test with the latest Marlin bugfix and report back here.

marcus-e-e, 2023-12-21:
Im still new to this but manage to build the latest merlin bugfix (cant get the screen working right).
When inserting the SD I still get the media init failure and the endstops stop working. Making the head crash into the side when homing. 

(I have decided to move away from this board since I have spent weeks trying to get it working without success. I still have it installed a am happy to continue testing, because I know a lot of people are struggling with this board and I want to help improving this all ready great initiative )
[Configuration_Ender3v2_merlin_bugfix_21x.zip](https://github.com/mriscoc/Ender3V2S1/files/13744335/Configuration_Ender3v2_merlin_bugfix_21x.zip)

mriscoc, 2023-12-21:
Can you confirm if the endstops are inverted with the current Professional Firmware sources?

marcus-e-e, 2023-12-21:
No. Starting the printer in a state where no endstop is triggered. All are "open".

Inserting an SD-card causes:
z to change to "triggered" and be unresponsive.
x to stay "open", but turns unresponsive. 
y to stay "open" and is still working.

mriscoc, 2023-12-23:
So the pin definition for endstops are wrong.

colgateam, 2023-12-23:
Just as an update, I never got this board working consistently and just got an e3v3 mini and moved to klipper.

marcus-e-e, 2023-12-28:
Without inserting an SD the endstops are working correct. 

Marlin has created a fix for this problem in this pull request: https://github.com/MarlinFirmware/Marlin/pull/26562
I can also confirm that the card now works with Merlin. SD now reads the files and endstops are also correct. 

Hope you can find some time to add this fix.

marcus-e-e, 2023-12-29:
Hi hacked the changes from the pull request mentioned above, into the Professional firmware. I dont really know what I am doing, but I got the SD card reading normally and the endstops works fine. Homing worked fine. I have not yet trammed and done a test print. Will do so when I have time.

mriscoc, 2024-01-08:
New sources released, also the latest changes in Marlin bugfix where applied.

marcus-e-e, 2024-01-08:
I had a quick try but could not get it to compile: 
`> platformio run -e STM32F401RE_freeruns
Processing STM32F401RE_freeruns (board: marlin_STM32F401RE_freeruns; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Processing Professional Firmware requirements
Error: can't detect the correct architecture`

mriscoc, 2024-01-08:
@marcus-e-e thank you, it is fixed now.

marcus-e-e, 2024-01-10:
The problems looks to be fixed.
SD card is read ok. 
Endstops works correct, even after entering an SD card. 
Homing worked fine. 
(I have not tried bed leveling and printing, but guess that will be fine.)

thanks allot @mriscoc !

mriscoc, 2024-01-11:
@marcus-e-e can you post your configuration files?

marcus-e-e, 2024-01-11:
I used the following config files. 
[Configuration_freeruns.zip](https://github.com/mriscoc/Ender3V2S1/files/13907672/Configuration_freeruns.zip)

mriscoc, 2024-01-12:
Thanks I uploaded the corresponding json to include the support for that board in the Configurator: https://github.com/mriscoc/Special_Configurations/commit/484d0510a3e9ab118352a2f3383478fe94ed635a

closed 2024-01-12 by mriscoc