mriscoc/Ender3V2S1 #1081: COmpile Error

gsuresh2u, 2023-09-06:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

Hi, when i try to compile my own firemware with your source for SKR Mini E3 V3, its giving error `"\marlin\src\lcd\e3v2\proui\proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this s
[Configs.zip](https://github.com/mriscoc/Ender3V2S1/files/12533686/Configs.zip)
cope"`

### Bug Timeline

form release Ender3V2S1-20230904

### Expected behavior

Need to compile

### Actual behavior

Not compiling

### Version of Professional Firmware

20230904

### Printer model

Ender 3V2

### Electronics

SKR Mini

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-09-06:
Update your configuration files with https://github.com/mriscoc/Special_Configurations

gsuresh2u, 2023-09-06:
Sir, i am using source files from https://github.com/mriscoc/Special_Configurations/tree/main/Ender3V2-SKRME3V3-BLTUBL-MPC

mriscoc, 2023-09-06:
DEF_PROBEZFIX is defined there: https://github.com/mriscoc/Special_Configurations/blob/b2e0180859c75e17d8fafd07c2c740a50691c926/Ender3V2-SKRME3V3-BLTUBL-MPC/Configuration.h#L3390

Travois123, 2023-09-06:
I had this same issue. Ender 3 v2 with an ezabl, 4.2.2 board. Came here looking for a solution after beating my head against the wall for 3 days. Added in the line of code that mriscoc posted in the previous comment and it compiled amd am now happily printing again. TYSVM!

Viperuss2009, 2023-09-07:
> Sir, i am using source files from https://github.com/mriscoc/Special_Configurations/tree/main/Ender3V2-SKRME3V3-BLTUBL-MPC

compiled successfully, need to download special configuration from page itself, not form releases if that makes sense 
[image]

mriscoc, 2023-09-07:
Pre-made configurations files have been deleted, from now, the proper way to have correct configuration files is by using the Configurator.  
  
[image]

closed 2023-09-07 by mriscoc

gsuresh2u, 2023-09-08:
Sir, sorry for late reply, still getting same error, from where can i download source files?
i downloaded from https://github.com/mriscoc/Ender3V2S1

mriscoc, 2023-09-08:
> from where can i download source files?

You must use the Configurator to generate the configuration files and firmware sources from the main repository.

gsuresh2u, 2023-09-08:
Yes, I used configurator, still same error

mriscoc, 2023-09-08:
> Yes, I used configurator, still same error

Look for DEF_PROBEZFIX in your configuration files, if you don't find it, you have old configuration files and you should try downloading the files again until you get DEF_PROBEZFIX.

gsuresh2u, 2023-09-08:
I found the files, when i try to use configurator, now its giving error

[image]

mriscoc, 2023-09-08:
You need to download **all** files from the [special configurations repository](https://github.com/mriscoc/Special_Configurations).

gsuresh2u, 2023-09-08:
Ok, Thanks