mriscoc/Ender3V2S1 #1082: [Feature Request] Consider adding a "reprobe" button to tramming wizard

Luro02, 2023-09-07:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

While using the Tramming-Wizard, I do something like this:
1. Execute Tramming Wizard
2. Look at screen and adjust the springs
3. Press Continue
4. Press on Tramming Wizard again...

I think it would be worthwhile to add a button to the tramming-wizard menu, to probe all points again without exiting to the main menu. I tried to do this myself, but the source code is quite confusing (I found 3 or 4 code pieces that look like they draw the tramming wizard menu? My guess is that you have multiple variants of the same code for different displays?)

### Expected behavior

```
    O              O
    O              O

[Reprobe]      Continue
```
Ideally it would select the `Reprobe` button, because it is the likeliest one to be pressed.

### Actual behavior

```
    O              O
    O              O

[Continue]
```

### Version of Professional Firmware

20230904

### Printer model

Ender 3 S1 F4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

TheWebMachine, 2023-09-07:
I believe the Manual Tramming menu is what you actually want; allowing you to recheck individual corners, one at a time, as you adjust them. I use the Wizard to start and get an idea of how out of whack the corners are *relative to each other*, then Manual Tram on each corner as I dial them in, then wizard at the end to confirm all is finally level.

Luro02, 2023-09-07:
I have never done it this way. So to confirm:

1. Execute the tramming wizard
2. Open the tramming menu where you can do manual tramming.
3. By pressing the buttons it moves the nozzle over the specified corner like with the z-offset
4. Then you adjust the knobs until something with a consistent height like a paper barely fits under it? => all corners should have the same distance?

(will try that later today)

mriscoc, 2023-09-07:
Adding a button, to save one click, will increase the size of the code more of 300 bytes, which means I will have to remove some feature for printers with F4 and RCT6 SoC.

Luro02, 2023-09-07:
Oh okay, then I will close this.

closed 2023-09-07 by Luro02