mriscoc/Ender3V2S1 #1085: Issue with 3D Touch bed leveling sensor

wokkeltje13, 2023-09-07:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I installed a 3D Touch sensor, loaded the latest firmware (Ender3V2-422-BLTUBL-MPC-20230904) and did a factory reset.
Strange thing happens after auto home that the nozzle goes up so even bed tramming is not working since the probe is not touching the bed. 

I followed the procedure (https://www.webcarpenter.com/blog/162-3D-Print---How-to-calibrate-Z-offset-with-a-BLTouch-bed-leveling-probe-sensor) as described and my Z offset is -7.88. but after homing the nozzle remains to high.

I do not expect this to be a bug but probably something I did not do correct.

Printer is Ender 3V2 and I used the mount from the procedure (https://www.thingiverse.com/thing:4605354) so offset should be 1.7, so my nozzle seems to be moving 6mm to high. 

Any advise?

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20230904

### Printer model

Creality Ender 3V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

SD Card (headless)

mriscoc, 2023-09-08:
The current version doesn't have issues with 3DTouch. Check your hardware and wiring.

wokkeltje13, 2023-09-09:
Think disabling the HA setting solved the issue

closed 2023-09-09 by wokkeltje13