mriscoc/Ender3V2S1 #1086: [BUG] (bug summary)

bombardeo38, 2023-09-08:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

en el manual traming elijas la esquina tato delantera como trasera solo mide las de la izquierda no se desplaza a los puntos derechos de la cama a parte el primer movimiento es extremadamente lento en el traming automatico si se desplaza y a buena velocidad

### Bug Timeline

a sido despues de actualizar a la ultima version

### Expected behavior

que se desplaca a las esquinas manualmente cuando las elijo 

### Actual behavior

cuando seleccion la esquina derecha no se desplaza sube y baja en la esquina izquierda

### Steps to Reproduce

ir a manual tramming y cambiar de izquierda a derecha da igual sies posterior o delantera

### Version of Professional Firmware

2.1.3 20230904

### Printer model

ender 3 v2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

mriscoc, 2023-09-09:
Did you set the correct mesh inset?  https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

bombardeo38, 2023-09-11:
si lo hice y calibre los pasos de motores fue un fallo puntual solo me a ocurrido 2 veces es como si la impresora no reconociera la posicion que tu le indicas de todas maneras si me puedes ayudar quiero configurar el gcode se salida para que nivele la cama ante de imprimir puse el g29 pero no hace caso en prusa ni cura ultima version gracias por todo

mriscoc, 2023-09-11:
https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts

Not a firmware issue

closed 2023-09-11 by mriscoc