mriscoc/Ender3V2S1 #1094: [BUG] Speaker will make initial double beep on start, may work when on but will stop working if double beep is called again

SargeR33, 2023-09-16:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Very strange. I was running an August release with no issues at all. Ender 3 v2 with 4.2.2 ret6 and a DACAI screen. Screen previously flashed with DACAI firmware, all working correctly.

Since updating to latest release with a custom build, the internal speaker will make its initial double beep on startup. It will work like normal until the system executes another double beep, it makes a odd double beep sound and then the speaker will stop working. It will no longer beep when navigating the menu and clicking with single beeps.

Its nit picky yes but, strange behavior. 

### Bug Timeline

After latest FW build and update.

### Expected behavior

Single beep while navigating through menu, double beep for store and other specific functions. 

### Actual behavior

Odd sounding double beep which will cause the speaker to no longer work until the machine is rebooted. It will however still produce double beeps that almost sound like they clip.

### Steps to Reproduce

Turn printer on, 50 50 chance the speaker will immediately stop. If it does not stop, going into any menu and hitting store settings has so far produced 100% situation where speaker will no longer work while navigating the menus. Speaker was enabled in FW build.

### Version of Professional Firmware

20230904 

### Printer model

Ender 3 V2

### Electronics

4.2.2 RET6

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-09-18:
Tested in precompiled
[image]
I can't reproduce the issue.

Crazy-Charles, 2023-09-20:
Can confirm that i have same issue on my custom builds with speaker enabled, straight after re-flashing and entering a menu it produces a single bleep when the encoder is pressed but as soon as i "save settings" the single bleep ceases to work, double bleeps still work ok

Crazy-Charles, 2023-09-20:
As a quick work around, change encoder.cpp to use buzzer.tone(duration, frequency), i used 50ms and 1000Hz, instead of buzzer.click(10)

SargeR33, 2023-09-23:
My mistake. I did NOT use a precompiled build. It was custom builds.

mriscoc, 2023-12-18:
Fixed in the latest ANNIVERSARY 2023 version

closed 2023-12-18 by mriscoc