mriscoc/Ender3V2S1 #1095: [BUG] Latest Source code won't compile with configuration from configurator)

carpe-zz, 2023-09-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When downloading the source code and using the output from your configurator tool, I get the below error when trying to compile.

### Bug Timeline

September release - 20230904

### Expected behavior

I've used the configurator for the past three releases with no issues compiling.
Selecting:

Ender3V2, 422, BLT (UBL Checked,) DACAI, T13, LA and MPC

### Actual behavior

platformio run --silent -e STM32F103RE_creality
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\MinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\HAL\STM32\MinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\msc_sd.cpp:20:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\HAL\STM32\msc_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\usb_serial.cpp:27:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\HAL\STM32\usb_serial.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/proui.h:97:21: error: 'DEF_PROBEZFIX' was not declared in this scope
   97 |   float probezfix = DEF_PROBEZFIX;
      |                     ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/../lcd/e3v2/proui/dwin_defines.h:107,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\HAL\STM32\MarlinSerial.h:24,
                 from Marlin\src\HAL\STM32\HAL.h:34,
                 from Marlin\src\HAL\STM32\MarlinSPI.h:24,
                 from Marlin\src\HAL\STM32\MarlinSPI.cpp:27:
[... 142 lines cut ...]
Adding configuration files from your configurator tool.

### Version of Professional Firmware

Source files downloaded 9/17/23

### Printer model

Ender 3 V2 

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

classicrocker883, 2023-09-18:
it doesnt appear DEF_PROBEZFIX is used. simply comment out any instances 

especially **lcd/e3v2/proui/proui.h** line : 97

mriscoc, 2023-09-18:
You need to update the [special configurations repository](https://github.com/mriscoc/Special_Configurations/archive/refs/heads/main.zip) (all files)

carpe-zz, 2023-09-19:
> You need to update the [special configurations repository](https://github.com/mriscoc/Special_Configurations/archive/refs/heads/main.zip) (all files)

Ahhh, this was it.  Somehow I had grabbed the special configurations sponsored zip instead.  My apologies, it's compiling without issue now!  Thanks

closed 2023-09-19 by carpe-zz