mriscoc/Ender3V2S1 #1096: [BUG] Cr Touch doesn't seem to be compensating enough

gergthewerg, 2023-09-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

i have the ender 3 v2, a cr touch, and a pei spring steel sheat on my bed. but the leveling doesnt seem to be enough. i can feel and see the z axis move, and the icon is blinking. yet some parts are waaay too close to the nozzle, and others are waaay to far. im also using the most recent version. this has also been happening with the old stock glass bed, but less so. 

heres my most recent start gcode:
`M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160 ; start warming extruder to 160
G28 ; Home all axes
G29 A ; Activate the UBL System.
G29 L0 ; Load UBL at slot 0
G29 J2 ; 4-point level
G29 F10.0 ; Fade to 10mm
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
M300 ; Printing has started`

### Bug Timeline

ever since i first put in my cr touch (and the fw, both were at the same time)

### Expected behavior

I expect the printer to compensate for the bed not being flat

### Actual behavior

its not doing that enough

### Steps to Reproduce

install cr touch
install custom fw
start print
watch print fail

### Version of Professional Firmware

SEPTEMBER 2023

### Printer model

Ender 3 v2

### Electronics

4.2.2 

### Add-ons

cr touch. pei coated steel sheet. spider 2.0 hotend

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

classicrocker883, 2023-09-18:
i also have a CR touch and never experienced any issues. if you can give some more detail or a video or something that can provide some clues as what is going on. when you did the auto bed leveling, how does your mesh look?

anything like this?
<p>[image]</p>

a difference of +/-0.1 like shown here doesnt actually really require bed leveling. I mean I could print with or without bed leveling enabled and there is little difference I can notice.

have you tried printing without bed leveling enabled?

gergthewerg, 2023-09-18:
> Currently there isn't leveling issues with the CR-Touch, be sure that you follow the guidelines about disabling HS.

Right, I forgot to mention that. I have disabled that too

gergthewerg, 2023-09-18:
> i also have a CR touch and never experienced any issues. if you can give some more detail or a video or something that can provide some clues as what is going on. when you did the auto bed leveling, how does your mesh look?
> 
> anything like this?
> 
> [image]
> 
> 
> 
> a difference of +/-0.1 like shown here doesnt actually really require bed leveling. I mean I could print with or without bed leveling enabled and there is little difference I can notice.
> 
> have you tried printing without bed leveling enabled?

No there's like huge differences, but like, in the middle of the bed. Although if I understand correctly, it should be within tolerances

Fallenleader, 2023-09-19:
What is concerning to me is the fact that scrolling through closed issues this type of issue has cropped up multiple times over and over.
In some cases, user error was the cause, but I can't be the only one who had probe issues like this poster claims.
I am going to attempt this firmware again, but I suspect that if these issues are still cropping up and being dismissed as a non issue, that the probing problem I had before will likely happen again.
I am already well aware I have a slightly warped bed, and am in the process of considering my options on how to compensate for the severe warps, but even with those the tramming results I got from stock hit far more accurate than this CFW did last I tried. I don't mean this as an insult, just clarification that something was indeed amiss (quite frankly I loved how much control I had with this CFW).
I would strongly recommend considering this as more than a non issue considering the multiple reports over time of this same problem, or a similar variations of it.
For a bit of prior history, my last usage of this firmware resulted in probing that in one instance would look like the image above, severely fail on a print job, and then become littered with +- 0.6 to +-0.8 readings. This can't just be a me tripping thing, much less the other users experiencing this.
That said, without an image from the OP, I can't confirm their issue is the exact same as the one I had.

gergthewerg, 2023-09-19:
> What is concerning to me is the fact that scrolling through closed issues this type of issue has cropped up multiple times over and over. In some cases, user error was the cause, but I can't be the [...]

I've noticed that a few times, but most of the time I saw it was user error. I'll post some pictures a bit later when I get home

gergthewerg, 2023-09-20:
Already tried, made no change :/

mriscoc, 2023-09-20:
> Already tried, made no change :/
Update your original question to reflect the changes, your current start g-code script posted there is not correct.

gergthewerg, 2023-09-20:
Well to be fair, I did say it was the most recent one I tried. I've tried several, but I don't remember/have all the ones I have though off of the top of my head

Plopsi, 2023-09-21:
Hello,

I had also problems, but with an S1 - somehow probing is not working well - still investigating what it causes it but here are two points to test, even whle No.2 is very time consuming:

1. really get rid if G29 J - it messes the mesh up, at least for me, really bad, might be a result of the other probing issues.
2. Manually readjust every point in the mesh via edit mesh. Mine went from a +/-0,2 to a +/-0,5. especially in the diagonal from 0,0 to 9,9 the difference in the measured and adjusted values are huge, up to 0,3mm difference. Readjusting every point manually made the UBL work perfectly, and the print is nice and adhere starting from the first layer.

-- Additional Info --
I currently don't know what the cause is but in the FL(0,0) the resulting compensation is far to low (nozzle touched bed) and on BR(9,9) far to high (nozzle flies high). FR and BL are more around normal while FR also is more nozzle touching and BL is flying.
After readjusting all points I now have a absolutely nice print picture.

The problem starts already when tramming the bed. I used the wizard to level initially and then use the tramming per corner with probe and adjusted every corner to +/-0,01. When I then manually check every point vial manually tramming (paper/feeling gauge) It's already here the same picture 0,0 much too high, the nozzle nearly drills into the bed and 9,9 much to high, I could put 4 papers under the nozzle.
As said I have no clue why this happens, I don't think it's the probe, the measured values are repeatable for every X/Y-point. Also aready checked if the X-gantry is out of level but it's <1mm on the sides, same in different heights.

So for now I stick to manual level and meshing the bed - and backup the mesh :D). I wanted to try at the weekend to reinstall the original FW and let a test run if the leveling there is also off - before I installed the ProFW the ABL of the OrigFW worked very well even with an quite unleveled X-Gantry, so... don't know.

- Plops

crazy-oatmeal, 2023-09-22:
I too am having an issue with mesh compensation. I even dropped a corner by a mm and re-probed to make sure I could see it compensating. I see no evidence of ABL operating during print. I went so far as to install a SKR E3V3 to see if it was some weird issue with the GD MCU my 4.2.2 was running

Plopsi, 2023-09-22:
I guess I found the problem. I haven't tested it yet but it is most likeley that the bed "wobbles" a bit while moving (not loose, it just moves/tilts while moving form 0 to max) - I will try to move the probe to a Zero-Offest to Y and see if it helps:

https://www.printables.com/de/model/257460-ender-3-s1-zero-offset-cr-touch-mount

But might take a while before i can test it :)

I think the correct term for this behaviour is "probe location bias" (I found that term in the Klipper documentation :D)

--- Additional Info ---
It came to my mind while trying to analyse the delta of the UBL measure and after I googled up Y axis issues I came to that alternate mount. I will test it later this night or tomorrow and will tell you if it works.

I did a test-row over 5 measures, sadly I forgot to save the last 3 after the Y axis problem came into my mind - I was to excited that this might be the solution :D But here are the result of the first 2 (used a 3x3 mesh for testing)

[image]

Left is UBL mesh, middle is the mesh after manual adjusting via papertest, third is the difference (Manual-UBL) and the further right outer is the difference of the difference between the two - which you can say is 0,0 (papertest darivation ;))

Also to note that in the first row I used the corner-probing to bring the tramming to 0,00 +/-0,01. as you see the problem already existed there, and afterwards UBL measured a nice - but wrong - mesh. The second row the bed was manually trammed with papertest.

-Plopsi

Plopsi, 2023-09-22:
First report. I printed and mounted the new Zero-Y-Axis-Probeholder and made a testmeasure.
Here is the measure before (I printed the part with this):

[image]

And here directly the first measure after the mount:

[image]

I think the pictures are telling their story. A first test with edit mesh shows that it seems the distance is now more or less consistent on the probingpoints.
Both are made with 240 ZFeed, 1 probe per point. No change in tramming, bed was not moved - while it had a bit of glue in the center on the second measure due to the previous print.

I now create a 9x9 mesh with 120feed and 3 probes per point and then will do a first one layer testprint.

- Plopsi

mriscoc, 2023-09-22:
Maybe related to https://github.com/mriscoc/Ender3V2S1/issues/614

Plopsi, 2023-09-22:
> Maybe related to #614

Definitely. The different probemount helped. The test print looks very nice - not perfect but I did no finetuning yet.
If the problem is not a twisted Y axis - which could happen - I guess it will most likely start when the Y-axis POM wheels starting to get some wear.

- Plopsi