mriscoc/Ender3V2S1 #1099: [BUG] When using ABL, Multiple Probing of 2 results in 3 total probes. There is no way to do 2 probes only.

mdburkey, 2023-09-21:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

While playing around with bed leveling on this firmware, I have noticed that there is no way to actually achieve the classic "2 probe" (one fast/one slow) option with bed leveling. As implemented right now, the only options are a single probe and 3 or more probes. This appears to be a side effect of EXTRA_PROBING being restricted to a hard coded value of 1 with PROUI_EX enabled. 

At the least, I would think that updating the menu item to be "Extra Probes" or similar would make more sense than just "Multiple probe". Basically, it isn't logical for a Multiple Probe of 2 to result in 3 probes. 

As a side note to this, I have looked all through the code to see where the actual logic for the probing operation is handled when PROUI_EX is enabled. For the disabled case, the code was pretty easy to find, as was the "multi_probing" variable assignment, but for the life of me I can't find where it is actually being used, otherwise I'd see if I could fix things to where a simple dual probe option worked again. 

### Expected behavior

I would expect a Multiple Probing value of 2 to result in 2 probes, not 3.

### Version of Professional Firmware

2.1.3 (20230904)

### Printer model

Ender 3 Max Neo

### Electronics

4.2.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-09-21:
It is not a bug, the restriction is by design, check the SanityCheck.h file for all the restrictions:
```cpp
    #if MULTIPLE_PROBING == 1
        #error "MULTIPLE_PROBING must be 0 or more than 2."
    #endif
    #if MULTIPLE_PROBING > 0 && EXTRA_PROBING != 1
      #error "PROUI_EX requires an EXTRA_PROBING value of 1"
    #endif
```

mdburkey, 2023-09-21:
Yes, I was already aware of the sanity check and the requirements for EXTRA_PROBING, etc, but I would still consider the inability to do two probes to be an issue, if not a bug. Basically, I still prefer the classic fast touch/slow touch methodology and then averaging the two values over a single touch (which I find to be less accurate) and while 3 or more touches works, it gets even more obnoxiously slow. 

That being the case, I will continue to look at the code and see if I can re-implement this functionality myself.