mriscoc/Ender3V2S1 #1102: [BUG] (bug summary)

Thefern3750, 2023-09-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I just installed mriscoc on my E3V2 with BlTouch, I can move the Z axis with the "move" section but it's doesn't work for the auto home

### Bug Timeline

Now

### Expected behavior

It's expected to move down

### Actual behavior

It's not moving

### Steps to Reproduce

1. Install the firmware
2. Do auto home

### Version of Professional Firmware

Ender3V2-422-BLUTBL-MPC

### Printer model

E3V2

### Electronics

422

### Add-ons

BlTouch ; PEI ; Dual Z

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

Thefern3750, 2023-09-23:
When It's trying to auto home the Z axis, the bltouch probe move down and up with red light

classicrocker883, 2023-10-18:
BL touch connector may not be installed right. check the connections, make sure theyre in the correct positions, and look for any loose pins.

Thefern3750, 2023-10-18:
> BL touch connector may not be installed right. check the connections, make sure theyre in the correct positions, and look for any loose pins.

It's all gold

Thefern3750, 2023-10-18:
Good*

closed 2023-12-31 by github-actions[bot]