mriscoc/Ender3V2S1 #1104: [BUG] Neopixel strip only first LED works

Ripwords, 2023-09-27:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I wired up my NeoPixel LED strip (19LEDs) I checked before hand with an ESP32 and an example code to ensure all LEDs are working and they are, but after wiring up on my Creality 4.2.2 and using the special configurator i managed to get the first LED of the 19 working as expected, changing colors and everything but only one out of 19 is working. Not really sure why

### Bug Timeline

Today

### Expected behavior

all 19 leds to light up

### Actual behavior

first led lit up but not the rest

### Version of Professional Firmware

August2023

### Printer model

Ender 3V2 

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

closed 2023-09-28 by Ripwords

mriscoc, 2023-09-28:
Hi, I don't use Neopixel so it is difficult to test that functionality with the recent Marlin bugfix changes. If you found something to correct, please post here the changes.

Ripwords, 2023-09-28:
In the end one of the connectors was damaged probably during soldering. Once removed everything worked as expected.Thanks regardless for the response 😁