mriscoc/Ender3V2S1 #1114: [BUG] (bug summary) Ender3V2 BLtouch level issue

ELEANOR6589, 2023-10-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I experience intermittent issues with my printer not placing the nozzle at the correct height after the BLtouch probes the bed. I have tried several different things and I am not sure if it is an issue of user error or something else. I have checked all mounting points to ensure the probe isnt moving, ran z probe wizard/set z offset, ran a bed level mesh, added a piece  of startup code to include M420 S1 to ensure the mesh is included. What confuses me is that I can print one time and it works as intended but it will eventually have this issue again. Usually within 1 or 2 prints. The print nozzle consistently will hover about 5mm (give or take) above the bed and never position itsself correctly after I go through leveling the bed. The only thing I can do is let the printer begin its print and then manually adjust the z offset during the print for a consistent result. 

### Bug Timeline

Old issue. Started about a month ago.

### Expected behavior

I expect the nozzle to move to the appropriate height above the bed after it has been leveled and it runs the z home probe test.

### Actual behavior

The nozzle moves itsself to begin the straight line purge on the left hand side of the plate but never drops down to the plate. It consistently drops to about 5mm and then begins extruding.

### Steps to Reproduce

1. Load file in printer.
2. select the file I want to print.
3. Print begins and it will never drop to the build plate

### Version of Professional Firmware

version 2.1.3

### Printer model

Ender 3V2

### Electronics

4.2.2

### Add-ons

bl touch and creality direct drive extruder

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Cura

mriscoc, 2023-10-21:
Sorry, your issue can't be confirmed.

Pally20, 2023-10-24:
I got the same issue. Slicer is Prusa, Ender 3V2, HW 4.2.2

ELEANOR6589, 2023-10-24:
> I got the same issue. Slicer is Prusa, Ender 3V2, HW 4.2.2

Maybe we can find a solution. I have no idea what to do. Only thing i can figure is it is related to hardware. I have done all that I can to make it work. Im sick of trying at this point. The printer worked fine for a couple months then started doing this. So im not sure what it would be but I know im going through leveling, mesh setup, z offset setup and z homing just about every print now and it still doenst matter. It will always be at least a few mm too high but it usually goes about 5mm too high every print

brainpaincgn, 2023-11-16:
I got the same issue. Slicer is Cura, Ender 3S1.

ELEANOR6589, 2023-11-16:
> I got the same issue. Slicer is Cura, Ender 3S1.

Apparently its just us thats the problem LOL. Nvm the fact I swapped to different firmware that doesnt use the BLtouch and now it is staying level and has no issues staying where I want it to.

VREvolver, 2023-11-17:
> > I got the same issue. Slicer is Cura, Ender 3S1.
> 
> Apparently its just us thats the problem LOL. Nvm the fact I swapped to different firmware that doesnt use the BLtouch and now it is staying level and has no issues staying where I want it to.

Hey! I'm having this issue too...

I had Professional Firmware from april (or so) and it was working perfectly, but then i updated it and now i'm having that issue as well. Funny thing is that i have two Ender3 v2 with sprite extruder and bl touch and both of them are having the same issue since i updated them!

VREvolver, 2023-11-17:
Oh, i figured it out... 

I installed JyersUI with HS and the issue kept hapening, so i installed it's version without "HS" on it, which fixed the probe issue, so i went back to MRISCOC firmware and disabled that option:

Advanced > Probe Settings > Enable HS Mode (unmark it)

I hope it fixes your issues as weel!

ELEANOR6589, 2023-11-17:
> Oh, i figured it out...
>
> I installed JyersUI with HS and the issue kept hapening, so i installed it's version without "HS" on it, which fixed the probe issue, so i went back to MRISCOC firmware and [...]

Thanks for the input. I never knew about that. I too installed the Jyers software.  but with no bed level sensor. I have messed with it until im tired of taking it apart to swap the sensors around and all that jazz. Hopefully ill find the motivation to go and swap back to MrIscoc later. We shall see. Im about to buy a Bambu Labs X1C anyway so who knows how I will feel about it after this LOL

Pally20, 2023-11-20:
> Oh, i figured it out...
>
> I installed JyersUI with HS and the issue kept hapening, so i installed it's version without "HS" on it, which fixed the probe issue, so i went back to MRISCOC firmware and [...]

Thanks a lot, I turn it off and everything is fine now :)

closed 2024-01-31 by github-actions[bot]