mriscoc/Ender3V2S1 #1120: [BUG/Issue] Changing the maximum hotend temperature

luisbrandao, 2023-10-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hello,

I upgraded my printer to an all metal hotend because i started using recycled PET filament and it need around 260~270 nozzle temperature range.

I been printing in the maximum allowed by default (260) but i need a little more, and dont know how to archive it.

I just discover the custom code:
`C104 U1 Tn`
and sendt it with octoprint. The printer acknowledge it, saying the new maximum is 275, but stop at 260 whenever i try anything past 260.

[image]

### Expected behavior

C104 U1 275 allowed me to heat it up to 275

### Actual behavior

C104 U1 275 does nothing

### Version of Professional Firmware

2.1.3 - 20230522 - 21:01

### Printer model

Creality Ender S1

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

luisbrandao, 2023-10-25:
Is there another alternative? i am not comfortable enough to try to build this firmware myself

mriscoc, 2023-10-27:
275 - 15 (HOTEND_OVERSHOOT) = 260

mriscoc, 2023-10-27:
> ### Actual behavior
> C104 U1 275 does nothing

Also, the correcto command format is
```gcode
C104 U1 T280
``` 
You missed the "T"

luisbrandao, 2023-11-06:
doesn't  seen to be it, adding the T didnt worked. I even tried to place it in the pre-gcode in cura (C104 U1 T300) , but it still setting to 260 anything beyond 260.

luisbrandao, 2023-11-10:
i didnt think it was a new feature.
the new firmware solved it, thanx

closed 2023-11-10 by luisbrandao