mriscoc/Ender3V2S1 #1123: [BUG] MINTEMP errors when trying to heat nozzle

cartag, 2023-10-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Whenever I try to heat my nozzle, I get a message of Printer Killed, Err: MINTEMP Turn off the printer. This only happens with this firmware, either precompiled or custom build. If I flash a traditional Marlin build it will heat fine. 

Receiving this error in Octoprintwhen I try to heat the nozzle:
MINTEMP triggered, system stopped! Heater_ID: 690 - T:26.07 /0.00 B:65.47 /0.00 @:0 B@:0

### Bug Timeline

New, fresh install

### Expected behavior

I expected the nozzle to heat without producing an error. 

### Actual behavior

The moment I select Preheat PLA it gives a MINTEMP error. Same thing happens if I try to pre heat via OctoPrint.

### Steps to Reproduce

1. Flash firmware
2. Select Advanced > Hotend PID Settings > Hotend PID (Nozzle heats fine here) 
3. Select Prepare >  Preheat PLA
4. Printer immediately starts fast beeping, gives error that Nozzle Temp is too low then loads a second error with a long sustained beep stating Printer Killed, Err: MINTEMP Turn off the printer 

### Version of Professional Firmware

2.1.3

### Printer model

Creality Ender 3V2

### Electronics

Board 4.2.2

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

den-ali, 2023-10-30:
same problem here so i Follow.

keith20mm, 2023-11-01:
Ditto... Went back to Creality OEM

jonnsoft, 2023-11-02:
I have the same problem.  Sometimes it works after a soft reboot.

tigwey, 2023-11-08:
Same issue for similar configuration, with T13 thermistor

mriscoc, 2023-12-18:
The MINTEMP error for the hot-end is usually caused by issues with the thermistor or missing heating block silicone jacket.

cartag, 2023-12-20:
This is only happening with this firmware though. I flashed the same printer with a generic Marlin build and it is next to me printing as I type this. If it were occurring with all firmware I'd agree with you that it would be a generic support request. I've not tried the latest version of the firmware but will try it this weekend.

sg84, 2023-12-22:
I've seen this too - latest release for Ender 3v2 BLTouch MPC. I found that after running the hotend PID tuning I kept getting the heating error too. If you're fortunate and have an EEPROM backup of your PID settings from the previous run, restoring them should let you print again on this firmware release.
At a total guess, there's a clash between the PID and MPC settings? All I've found is that restoring my PID settings from a previous EEPROM backup fixes it.

closed 2024-03-03 by github-actions[bot]