mriscoc/Ender3V2S1 #1124: [BUG] (bug summary)

PedanticAvenger, 2023-10-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

MPC Autotune fails out with MINTEMP error.

Selecting preheat for any filament type works and heats up as expected.

### Bug Timeline

Testing custom build of current.

### Expected behavior

Autotune to run as normal.

### Actual behavior

MPC Autotune is acting like it is ignoring PREHEAT_TIME_HOTEND_MS VALUE.

Autotune process checks for ambient, then starts the heat cycle and almost immediately errors.  
The temp sensor is T66 type (reason for custom, all other options are normal as per configurator script).
PREHEAT_TIME value is set to 30000

### Steps to Reproduce

Boot up printer
Start MPC Autotune
ERror happens

### Version of Professional Firmware

2.1.3   20231028

### Printer model

3v2

### Electronics

4.2.2

### Add-ons

MicroSwiss NG, Copper heat block w/ T66 temp sensor and 50W heater cartridge

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Additional information & file uploads

Didn't send anything from slicer yet or from octo.
This is a migration from Jyers which was working.

mriscoc, 2023-11-01:
Try disabling MPC and using PIDTEMP, also disable HAS_PROUI_MAXTEMP for better compatibility with non standard thermistors.

mriscoc, 2023-11-01:
Check your issue title for remove "invalid" label.

lawhitecross, 2023-12-17:
im using the SKR Mini V3 and im getting the same error after 2 hours of print and then the print fails. I am using the precompiled version.
Printer reports firmware name "Marlin 2.1.3 MRiscoC Ender3V2-SKRME3V3-BLTUBL-MPC, based on bugfix-2.1.x (Sep  4 2023 10:09:37)"

mriscoc, 2023-12-19:
@lawhitecross currently the firmware is build only for T1 (stock) and T13 thermistors, if you have the T66 as indicated in the title of this post, you need to compile a version specific for that thermistor with HAS_PROUI_MAXTEMP disabled for better compatibility.

closed 2023-12-19 by mriscoc