mriscoc/Ender3V2S1 #1126: [BUG] Control knob randomly skips or increases/decreases arbitrary numbers on Ender 3 V2

TheRealOne78, 2023-10-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Scrolling or changing values is a pain because the firmware makes up some random values or movements, sometimes it's so bad, moving just 1 step adds up for example 100 more steps.

The worst is when for example, you are rotating the knob to the right and it reads it as left and viceversa.

### Bug Timeline

Kind of old. However some firmwares did not have this issues. (I don't remember which and can't test, because release assets are missing)

### Expected behavior

Scroll as expected.

### Steps to Reproduce

1. Flash https://github.com/mriscoc/Ender3V2S1/releases/download/20230904/Ender3V2-422-BLTUBL-MPC-20230904.bin
2. Play around with the knob by selecting a value and adjusting it.

### Version of Professional Firmware

20230904 (Latest as I write this issue)

### Printer model

Ender 3 V2

### Electronics

4.2.2

### Add-ons

I installed a CR-Touch.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

Video showing the problem: https://therealone78.ddns.net/uploads/control_knob.webm
_Note_: Sometimes it's even worse, being unable to set for example the Z-Axis while printing.

TheRealOne78, 2023-10-29:
Note: I don't think it's a hardware issue, since I don't experience this with the official Creality firmware and I didn't experience this in earlier versions of this firmware (I am unable to re-flash older firmwares since they are not available in the release assets anymore)

classicrocker883, 2023-10-29:
changing `#define ENCODER_WAIT_MS` from **lcd/e3v2/common/encoder.h** has made encoder issues better, from `20` to `10`.

youre right to say earlier versions did have this value at `10`, until recently it is `20`, which other official versions have.

TheRealOne78, 2023-10-30:
Changed the `ENCODER_WAIT_MS` value. Setting it higher seems to be a little bit better while setting it lower makes changing values faster. But even then, the numbers jump randomly. Say for example your value is 10 and you rotate by just one value, but it adds up over 100.

classicrocker883, 2023-10-31:
you can have an issue with the encoder itself...that is its possible a connection on the LCD board may be "loose" or needs to be reflowed. I would suggest try using CRC 2-26 electrical safe lubricant. I use it on anything, just spray this - it can help revive the knob encoder.

in **Configuration_adv.h** you will find line ~1470  
```y
#define ENCODER_RATE_MULTIPLIER
#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #define ENCODER_5X_STEPS_PER_SEC    30  // (steps/s) Encoder rate for 5x speed
  #define ENCODER_10X_STEPS_PER_SEC   80  // (steps/s) Encoder rate for 10x speed  // Ender Configs
  #define ENCODER_100X_STEPS_PER_SEC 130  // (steps/s) Encoder rate for 100x speed  // Ender Configs
#endif
```
basically think of the values as how fast you turn the knob, the faster you spin the knob, the encoder then multiplies the value you want changed. so if you spin it at least 130 steps/s, it will multiply the value by 100x. say you revert the numbers around, the faster you spin the knob, the value wont change as much.

---
these are the values I use:

```r
40
70
110
```
but I also changed the values in **src/lcd/e3v2/common/encoder.cpp** line ~150  
```y
               if (encoderStepRate >= ENCODER_100X_STEPS_PER_SEC) encoderMultiplier = 135;
          else if (encoderStepRate >= ENCODER_10X_STEPS_PER_SEC)  encoderMultiplier = 25;
          #if ENCODER_5X_STEPS_PER_SEC
            else if (encoderStepRate >= ENCODER_5X_STEPS_PER_SEC) encoderMultiplier = 5;
          #endif
```

also, if you'd like to mess around with these values on the fly, I modified the code so I was able to change the encoder values right in the LCD menu. makes it way easier than having to recompile and reflash everytime....

mriscoc, 2023-11-01:
Your encoder has started to fail, you can mitigate that failure by enabling:
```
#define SMOOTH_ENCODER_MENUITEMS  // Menu items value faster/smooth change rate
```
In the configuration.h file.

TheRealOne78, 2023-11-01:
That looks like a good workaround. Might as well replace this encoder as soon as possible.

I have been not able to test this yet, but I will come with an answer as soon as I can.

Thank you for your help

TheRealOne78, 2023-11-01:
I just compiled it with `SMOOTH_ENCODER_MENUITEMS`, this is much better.Thank you.

I'll let you decide whether you want to close this issue or wait for me to replace the encoder and see the results (which wouldn't be too soon)

TheRealOne78, 2023-11-03:
I ended up replacing the encoder sooner than I expected.

closed 2023-11-03 by TheRealOne78