mriscoc/Ender3V2S1 #1128: [BUG] s1 issue, tramming no longer shows numbers

roadburp, 2023-11-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

firmware doesnt work, i cant abl, or tramm.

### Expected behavior

i expected it to tramm

### Actual behavior

its showing colors

### Steps to Reproduce

have a f4 board i guess

### Version of Professional Firmware

the f4 ubl version

### Printer model

f4 s1

### Electronics

?

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

[image]

mriscoc, 2023-11-06:
Already answered, please use the search function in the Github. Look for TJC

closed 2023-11-06 by mriscoc

roadburp, 2023-11-06:
i did install it, it changes nothing. it still shows squares.