mriscoc/Ender3V2S1 #1132: [BUG] (Display frozen after firmware flash)

rman215, 2023-11-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hello, 

I have an Ender 3 S1 with the STM32401 chipset. I've tried flashing the firmware in two ways. Firstly, by placing the file "Ender3S1-F4-20230904.bin" in the root of the SD card. The result from that attempt is that it ignored the bin file. So, I tried placing it in a folder named "STM32F4_UPDATE". This time, it appeared to recognize the firmware file, but the Creality logo stayed frozen on the screen. I left it for about 10 minutes before shutting the printer off, and the logo was still frozen on the screen.

It's possible that the firmware upgrade to the board went through, but the display firmware is not compatible. I then tried updating both the firmware on the board and the display, thinking that if it was running the latest firmware on the display, it might load correctly. However, that was not the case; I encountered the same issue again.

I would appreciate some help troubleshooting this issue. Thanks!

### Bug Timeline

New bug

### Expected behavior

After flashing the latest firmware, "Ender3S1-F4-20230904.bin" to chipset,  "STM32401" I should be welcome by the HUD with the firmware loaded in.  

### Actual behavior

After flashing the firmware to the main board the Creality logo stays frozen on screen. 

### Steps to Reproduce

1. Format SD card to FAT32, Allocated unit size 4096
2. Copy the file Ender3S1-F4-20230904.bin to root of SD card *read description tried two different methods
3. With the Ender 3 power off insert SD card
4. Power on printer and wait for Firmware flashing to complete

### Version of Professional Firmware

V3.2.0_Nice_C *Creality Firmware 

### Printer model

Creality Ender 3 

### Electronics

STM32401

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Additional information & file uploads

[image]

Crazy-Charles, 2023-11-12:
Have you read this?  https://github.com/mriscoc/Ender3V2S1/issues/966

rman215, 2023-11-13:
I hadn't seen this. As I suspected, it appears to be a firmware issue with the display. I'll try flashing 'Ender-3 S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER' and see if it makes a difference. Thanks!

rman215, 2023-11-16:
Rolling back the firmware did fix the issue. If anyone else is having a similar issue use release, "S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER."

closed 2023-11-16 by rman215