mriscoc/Ender3V2S1 #1141: [BUG] [DRAFT] Filament Management Cooldown doesn't turn off bed heater

TheWebMachine, 2023-11-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

NOTE: This is presently a draft, as I won't have time to fully diagnose this until tomorrow. Of course, I discover this on the busiest day of my week.

The summary: If you Preheat from main Prepare menu, it heats hotend and bed. If you then choose Cooldown from Filament Management sub-menu, it only turns off hotend, not bed. So bed will stay hot.

More testing needed to confirm. Give me 24hrs...

### Expected behavior

I expect Filament Management Cooldown to also turn off bed, even if Filament Management Preheat doesn't turn it on, as user may have preheated from previous main Prepare menu, which *does* preheat bed.

### Actual behavior

Filament Management Cooldown only turns off hotend, leaving bed in whichever state it was already in; on or off.

### Steps to Reproduce

TBD

### Version of Professional Firmware

latest (will update later)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-11-15:
That's behavior is by design, the **preheat hotend** item in filament management menu only has effect in the hotend temperature. So you can change the hotend temperature without lost the temperature stability of the bed.

TheWebMachine, 2023-11-15:
Oh...perhaps some clarification of menu option text to avoid confusion, then? I mistakenly left my bed heated for quite a long time (and burned myself when I returned 🤦🏻‍♂️) because it doesn't have an idle timeout. My brain presumed Cooldown to be a complete cooldown, regardless of menu. 🤷🏻‍♂️

closed 2024-01-15 by TheWebMachine