mriscoc/Ender3V2S1 #1145: [BUG] (Values don't display for Tramming or Bed mesh)

rman215, 2023-11-16:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When running the tramming/bed mesh wizard, the values do not appear in the grid display, only the colors do. Please watch the attached video for more details.

### Bug Timeline

New bug

### Expected behavior

Grid pattern should display measured values along with corresponding color. 

### Actual behavior

Only the color appears in the grid pattern when running the tramming/bed mesh wizard. 

### Steps to Reproduce

1. Click on Prepare from main menu
2. Select Bed Traming 
3. Choose Bed Traming Wizard

### Version of Professional Firmware

Ender3S1-F4-20230904

### Printer model

Creality Ender 3 S1

### Electronics

STM32401

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

https://github.com/mriscoc/Ender3V2S1/assets/122302689/d4040366-564d-4c8a-b45a-7ae7615c076d

mriscoc, 2023-11-16:
TJC

rman215, 2023-11-17:
I was reading through the original bug, but it's a little unclear. Do I need to compile special firmware for the TJC display? If so, is there a guide available to walk me through the process?

rman215, 2023-11-17:
Disregard the my last question. I found my answer here: https://github.com/mriscoc/Special_Configurations/releases/tag/tjc

closed 2023-11-17 by rman215