mriscoc/Ender3V2S1 #1150: [BUG] Auto build mesh, probe does not touch the bed

bulcke, 2023-11-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Installed BLtouch. It is working, as in: homing z works with the probe. I applied the Z offset using the wizard.

When using Auto build mesh, the printer "probes" all points and finishes. But in fact, the probe never touches the bed. Resulting obviously in a nonexistent flat mesh. It seems the gantry just does not go low enough.

Tramming wizard has the same behaviour, but at the first point (front left) after not touching the bed, it shows error "Probe out of bounds" on display.

To be clear, I repeat: when homing Z, the probe DOES touch the bed.

### Bug Timeline

just started using this FW

### Expected behavior

Probe actually touching the bed when probing when creating a bed mesh and tramming.

### Actual behavior

Probe not touching the bed because the gantry does not go lower.

### Steps to Reproduce

see bug description

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20230904.bin

### Printer model

Creality Ender 3V2

### Electronics

4.2.2 GD chip

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-11-19:
Check  de video at: https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#bad-3dblcrtouch-connection

Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

bulcke, 2023-11-19:
And I'm not asking for printer support.  I'm submitting a bug in your firmware. That's the proper thing to help open source projects and the community. And that's your answer? If you had read my bug report, you would have seen that the probe works fine.

I'm well aware of how 3D printers work or should work. I'm well aware of the features and limitations. I'm well aware of how a probe should work. I have enough experience to recognize a bug in firmware when I see it.

mriscoc, 2023-11-19:
Check your physical settings: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

Pally20, 2023-11-19:
I can confirm I have the same bug, I thing the last release is not good, I have lot of defects...

bulcke, 2023-11-19:
> Check your physical settings: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
> 
> You can use the [G30](https://marlinfw.org/docs/gcode/G030.html) G-code to debug and be sure that you are trying to probe reachable locations.

All mesh probing points are well within the build area.
And it's not that the build plate itself is way off. Trammed it manually, so it should be quite level with the frame. I did a few prints (obviously without mesh compensation) and they came out fine. First layer is ok.

Pally20, 2023-11-20:
> Check your physical settings: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides
> 
> You can use the [G30](https://marlinfw.org/docs/gcode/G030.html) G-code to debug and be sure that you are trying to probe reachable locations.

I unmark HS Mode and now it seem everything is fine :)

Advanced > Probe Settings > Enable HS Mode (unmark it)

bulcke, 2023-11-20:
Calibration guide only mentions "If you have a CRTouch check the notes about [disabling HS mode](https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch), if you intend to use a mesh larger than 5x5."

I'm not using a CRTouch and I am not using a mesh larger than 5x5.

I will try disabling HS mode tonight.

bulcke, 2023-11-20:
I can confirm with HS mode disabled bed meshing and tramming wizard work.

Bug or feature? ;-)

mriscoc, 2023-11-20:
> I can confirm with HS mode disabled bed meshing and tramming wizard work.
> 
> Bug or feature? ;-)

HS is an enhanced feature, some probes are incompatible with HS mode.

classicrocker883, 2023-11-25:
I think this is the main difference between the newer Marlin-bugfix. in **src/module/probe.cpp** line ~600
```diff
-// Ensure the BLTouch is deployed. Does nothing if already deployed.
-if (TERN0(BLTOUCH, bltouch.deploy())) return true;

+#if ENABLED(BLTOUCH)
+  // Ensure the BLTouch is deployed. (Does nothing if already deployed.)
+  // Don't deploy with high_speed_mode enabled. The probe already re-deploys itself.
+  if (TERN(MEASURE_BACKLASH_WHEN_PROBING, true, !bltouch.high_speed_mode) && bltouch.deploy())
+    return true;
+#endif
```

also there is an added `float probe_at_point()` function which isnt include in Marlin.