mriscoc/Ender3V2S1 #1154: Freeze Screen on booting

DA21bad, 2023-11-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hi, I was updating my firmware to the professional firmware from this repository, https://github.com/mriscoc/Ender3V2S1 . I have a 4.2.2 board but at the time I tried to flash it then the screen freezed.

[image]

please help me what can I do

### Expected behavior

I was expecting to initiate as common after a firmware update

### Actual behavior

Screen freezed

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20230904

### Printer model

Ender 3 V2 Neo

### Electronics

Board 4.2.2

### Add-ons

CR touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-11-26:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2023-12-04:
Some TJC displays need to be updated to boot properly, that is already explained in the Wiki.

DA21bad, 2023-12-05:
> Hi DA21bad <https://github.com/DA21bad>
>
> I experience exactly the same issue with screen freeze on one of my Ender
> 3 v2 Neo with BLTouch support and TJC screen (not sure if the screen
> somehow may impact here). Funny thing is that my another Ender 3 v2 Neo
> with exactly same board got MRiscoC without an issue and operated ideally.
>
> Have you by chance overcome this issue or yet you're unable to install
> MRiscoC?

Hi,

No I haven't tried again on updating the printer. I reflashed the factory
firmware, but I am willing to give it a chance again this week, I will
write again with the conclusions.

Kind Regards,
Diego

riccardopiola, 2024-01-05:
I had the same problem with my ender 3 v2 neo. I tried to update the display to the [latest firmware](https://raw.githubusercontent.com/mriscoc/Ender3V2S1/Ender3V2S1-Released/display%20assets/tjc_update.zip) provided in this repo but the update was not starting and the memory card became corrupted requiring a reformat when plugged in the pc.

To resolve you have to first update to the latest official creality display firmware https://www.creality.com/pages/download-ender-3-v2-neo, which in my case was "Ender-3 V2 Neo-Marlin2.0.8.3-HW-V4.2.2-SW-V1.1.8C-CR-Touch-20230324(9Langues)" following the instructions in the readme (you need to copy an extra file DC48270M043_XGUS.zlib other than the TJC_SET folder into the sdcard).

Then you can flash the [updated firmware](https://raw.githubusercontent.com/mriscoc/Ender3V2S1/Ender3V2S1-Released/display%20assets/tjc_update.zip) and it should work!