mriscoc/Ender3V2S1 #1156: [BUG] CRTouch/BLTouch always "triggered"

josurr, 2023-11-26:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The "Z-axis" endstop shows "Triggered" in the Endstops Diag screen in the UI.

This occurs on multiple v4.2.7 boards and multiple CRTouch probes with custom and pre-compiled (9/4/2023) firmware.

### Bug Timeline

Since building/flashing firmware

### Expected behavior

All axis home correctly.

### Actual behavior

When Auto-Homing, the X and Y axis home correctly, but when it comes to the Z axis, the probe deploys 3 consecutive times (without the print-head moving up or down) and does not stow after the third time (with the LED status on the probe staying illuminated blue).

The "status" on the stock display says "STOPPED" in the notification bar after the 3rd probe deployment.

### Steps to Reproduce

1. Turn on printer 
2. Open Advanced Settings screen
3. Scroll down to "Endstops diag" and open it
4. Observe that the Z-axis is "Triggered"
5. Go back to the home screen and activate the "Auto-Home" option
6. Observe X and Y axis home correctly and then the print-head moves to the homing point on the print bed
7. Observe the probe deploying 3 consecutive times (without the print-head moving up or down) and not stowing after the third time (with the LED status on the probe staying illuminated blue)
8. Observe the "status" on the stock display says "STOPPED" in the notification bar after the 3rd probe deployment

### Version of Professional Firmware

v2.1.3 20231126-153542

### Printer model

Creality Ender 3v2

### Electronics

4.2.7

### Add-ons

CRTouch bed leveling probe

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-11-27:
The end-stop diagnostic tool is only to check end-stop switches, not probes like CR/3D/BLtouch.

josurr, 2023-11-27:
The end-stop shows "Triggered" regardless of the probe being plugged into the board as well.

The version # I provided was from the build I made on my machine. The issue occurs on both custom builds _**and**_ the posted pre-built firmware on github (dated 9/4/2023)

mriscoc, 2023-11-28:
The printer is detecting an error with your CRTouch

```cpp
/**
 * Print an error and stop()
 */
void Probe::probe_error_stop() {
  SERIAL_ERROR_START();
  SERIAL_ECHOPGM(STR_STOP_PRE);
  #if ANY(Z_PROBE_SLED, Z_PROBE_ALLEN_KEY)
    SERIAL_ECHOPGM(STR_STOP_UNHOMED);
  #elif ENABLED(BLTOUCH)
    SERIAL_ECHOPGM(STR_STOP_BLTOUCH);
  #endif
  SERIAL_ECHOLNPGM(STR_STOP_POST);
  stop();   // <<<<< this shows the STOPPED message
}
```

josurr, 2023-11-28:
Ok, but this is with 2 different v4.2.7 boards and 2 different probes (including a brand new instance of both the board and probe).

Is there any way to get the specifics on what error it is detecting?

josurr, 2023-11-29:
Problem was with the line

`#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

Not being commented out (the CRTouch probe uses the 5-pin header on the controller board immediately next to the LCD header).

After commenting out that line, compiling, and flashing, the "Auto-Home" feature is working.

closed 2023-11-29 by josurr

mriscoc, 2023-11-29:
That is already commented out

[image]