mriscoc/Ender3V2S1 #1161: Weird OctoPrint Random Issue.

Moosiecakes, 2023-11-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Sent a Print from Cura to OctoPrint. Occasionally The display on the printer shows the Job is complete, Yet it hasn't started yet. Then after it heats the Nozzle & the bed it starts the print. Running down the edge of the bed and back Which is fine, Then it Moves onto the bed to start the print. The extruder does not move, The print head moves around like it's printing but the Extruder is not Moving, and no Filament is coming out of the Nozzle. 

### Bug Timeline

Only been using this Firmware for 3 days. Does it 1x a day.

### Expected behavior

Just print as normal. 

### Actual behavior

Print head moves around like its printing, No filament being extruded. 
Ender Display shows 100% complete job.

### Steps to Reproduce

Cannot reproduce, so far seems random. 

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20230904

### Printer model

Ender 3 V2

### Electronics

4.2.2

### Add-ons

Cr touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

Moosiecakes, 2023-11-30:
Today its done it now about 5 x..  Included is a screenshot. Taken Directly after the Job was sent to the Printer. 
[image]

mriscoc, 2023-11-30:
Look at Octoprint terminal for any error message.

Moosiecakes, 2023-11-30:
Ok here is the Terminal This is happening now almost every time i print. I have to either Cancel the job on Octoprint or on the actual printer. As show on the logs. 

Changing monitoring state from "Operational" to "Starting"
Send: N71933 M73 P100 R0*27
Recv: ok N71933 P15 B15
Send: N0 M110 N0*125
Recv: ok N0 P15 B15
Send: N1 M75 CE3E3V2_ABL Nut.gcode*44
Recv: ok N1 P15 B15
Changing monitoring state from "Starting" to "Printing"
Send: N2 M140 S50*82
Recv: ok N2 P15 B15
Send: N3 M105*36
Recv: ok T:24.43 /0.00 B:26.58 /50.00 @:0 B@:0
Send: N4 M190 S50*89
Recv:  T:24.43 /0.00 B:26.58 /50.00 @:0 B@:0 W:?
Recv:  T:24.49 /0.00 B:26.79 /50.00 @:0 B@:127 W:?
Recv:  T:24.49 /0.00 B:26.79 /50.00 @:0 B@:127
Recv:  T:24.46 /0.00 B:27.02 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.47 /0.00 B:27.39 /50.00 @:0 B@:127 W:?
Recv:  T:24.47 /0.00 B:27.39 /50.00 @:0 B@:127
Recv:  T:24.50 /0.00 B:27.76 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.43 /0.00 B:28.08 /50.00 @:0 B@:127 W:?
Recv:  T:24.43 /0.00 B:28.08 /50.00 @:0 B@:127
Recv:  T:24.43 /0.00 B:28.43 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.47 /0.00 B:28.87 /50.00 @:0 B@:127 W:?
Recv:  T:24.47 /0.00 B:28.87 /50.00 @:0 B@:127
Recv:  T:24.48 /0.00 B:29.26 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.46 /0.00 B:29.67 /50.00 @:0 B@:127 W:?
Recv:  T:24.46 /0.00 B:29.67 /50.00 @:0 B@:127
Recv:  T:24.49 /0.00 B:30.10 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.51 /0.00 B:30.53 /50.00 @:0 B@:127 W:?
Recv:  T:24.51 /0.00 B:30.53 /50.00 @:0 B@:127
Recv:  T:24.45 /0.00 B:30.96 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.49 /0.00 B:31.40 /50.00 @:0 B@:127 W:?
Recv:  T:24.49 /0.00 B:31.40 /50.00 @:0 B@:127
Recv:  T:24.50 /0.00 B:31.81 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.42 /0.00 B:32.29 /50.00 @:0 B@:127 W:?
Recv:  T:24.42 /0.00 B:32.29 /50.00 @:0 B@:127
Recv:  T:24.46 /0.00 B:32.78 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.51 /0.00 B:33.14 /50.00 @:0 B@:127 W:?
Recv:  T:24.51 /0.00 B:33.14 /50.00 @:0 B@:127
Recv:  T:24.45 /0.00 B:33.61 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.48 /0.00 B:34.03 /50.00 @:0 B@:127 W:?
Recv:  T:24.48 /0.00 B:34.03 /50.00 @:0 B@:127
Recv:  T:24.45 /0.00 B:34.52 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.47 /0.00 B:34.88 /50.00 @:0 B@:127 W:?
Recv:  T:24.47 /0.00 B:34.88 /50.00 @:0 B@:127
Recv:  T:24.46 /0.00 B:35.28 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.49 /0.00 B:35.68 /50.00 @:0 B@:127 W:?
Recv:  T:24.49 /0.00 B:35.68 /50.00 @:0 B@:127
Recv:  T:24.45 /0.00 B:36.03 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.44 /0.00 B:36.30 /50.00 @:0 B@:127 W:?
Recv:  T:24.44 /0.00 B:36.30 /50.00 @:0 B@:127
Recv:  T:24.42 /0.00 B:36.61 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.44 /0.00 B:37.01 /50.00 @:0 B@:127 W:?
Recv:  T:24.44 /0.00 B:37.01 /50.00 @:0 B@:127
Recv:  T:24.48 /0.00 B:37.29 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:24.44 /0.00 B:37.59 /50.00 @:0 B@:127 W:?
Recv:  T:24.44 /0.00 B:37.59 /50.00 @:0 B@:127
Recv:  T:24.47 /0.00 B:37.94 /50.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Send: N5 M108*47
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: ok N4 P15 B15
Recv: //action:notification Print Aborted
Recv: ok N5 P15 B15
Send: N6 M73 P0*81
Recv: ok N5 P15 B15
Send: N7 M117 ETL 01h34m19s*6
Recv: ok N6 P15 B15
Recv: ok N7 P15 B15
Send: N8 M73 P0*95
Send: N9 M117 ETA 11/30 1.52PM*116
Recv: ok N8 P15 B15
Send: N10 M73 P0*102
Recv: ok N9 P15 B15
Send: N11 M117 Print Done*101
Recv: ok N10 P15 B15
Send: N12 M73 P100*101
Recv: ok N11 P15 B15
Send: N13 M77*33
Recv: ok N12 P15 B15
Recv: ok N13 P15 B15
Send: N14 M117 Print Done*96
Recv: ok N14 P15 B15
Send: N15 M73 P100*98
Recv: ok N15 P15 B15
Send: N16 M117 Cancelling*25
Recv: ok N16 P15 B15
Send: N17 G27*42
Recv: echo:Home XYZ First
Recv: ok N17 P15 B15
Send: N18 M84*38
Recv: ok N18 P15 B15
Recv:  T:24.47 /0.00 B:38.30 /50.00 @:0 B@:127
Send: N19 M104 T0 S0*25
Recv: ok N19 P15 B15
Send: N20 M140 S0*87
Recv: ok N20 P15 B15
Send: N21 M106 S0*84
Recv: ok N21 P15 B15
Send: N22 M117 Print was cancelled*107
Recv: ok N22 P15 B15
Changing monitoring state from "Cancelling" to "Operational"

Moosiecakes, 2023-11-30:
This is Serial.log from octoprint. 

2023-11-30 12:17:46,635 - Changing monitoring state from "Operational" to "Starting"
2023-11-30 12:17:46,779 - Send: N71933 M73 P100 R0*27
2023-11-30 12:17:46,788 - Recv: ok N71933 P15 B15
2023-11-30 12:17:46,794 - Send: N0 M110 N0*125
2023-11-30 12:17:46,809 - Recv: ok N0 P15 B15
2023-11-30 12:17:46,811 - Send: N1 M75 CE3E3V2_ABL Nut.gcode*44
2023-11-30 12:17:46,826 - Recv: ok N1 P15 B15
2023-11-30 12:17:46,831 - Changing monitoring state from "Starting" to "Printing"
2023-11-30 12:17:46,846 - Send: N2 M140 S50*82
2023-11-30 12:17:46,860 - Recv: ok N2 P15 B15
2023-11-30 12:17:46,863 - Send: N3 M105*36
2023-11-30 12:17:46,876 - Recv: ok T:24.43 /0.00 B:26.58 /50.00 @:0 B@:0
2023-11-30 12:17:46,885 - Send: N4 M190 S50*89
2023-11-30 12:17:46,901 - Recv:  T:24.43 /0.00 B:26.58 /50.00 @:0 B@:0 W:?
2023-11-30 12:17:47,954 - Recv:  T:24.49 /0.00 B:26.79 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:47,969 - Recv:  T:24.49 /0.00 B:26.79 /50.00 @:0 B@:127
2023-11-30 12:17:48,913 - Recv:  T:24.46 /0.00 B:27.02 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:48,936 - Recv: echo:busy: processing
2023-11-30 12:17:49,919 - Recv:  T:24.47 /0.00 B:27.39 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:49,923 - Recv:  T:24.47 /0.00 B:27.39 /50.00 @:0 B@:127
2023-11-30 12:17:50,897 - Recv:  T:24.50 /0.00 B:27.76 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:50,919 - Recv: echo:busy: processing
2023-11-30 12:17:51,951 - Recv:  T:24.43 /0.00 B:28.08 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:51,955 - Recv:  T:24.43 /0.00 B:28.08 /50.00 @:0 B@:127
2023-11-30 12:17:52,895 - Recv:  T:24.43 /0.00 B:28.43 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:52,908 - Recv: echo:busy: processing
2023-11-30 12:17:53,897 - Recv:  T:24.47 /0.00 B:28.87 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:53,906 - Recv:  T:24.47 /0.00 B:28.87 /50.00 @:0 B@:127
2023-11-30 12:17:54,894 - Recv:  T:24.48 /0.00 B:29.26 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:54,903 - Recv: echo:busy: processing
2023-11-30 12:17:55,896 - Recv:  T:24.46 /0.00 B:29.67 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:55,906 - Recv:  T:24.46 /0.00 B:29.67 /50.00 @:0 B@:127
2023-11-30 12:17:56,894 - Recv:  T:24.49 /0.00 B:30.10 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:56,903 - Recv: echo:busy: processing
2023-11-30 12:17:57,896 - Recv:  T:24.51 /0.00 B:30.53 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:57,905 - Recv:  T:24.51 /0.00 B:30.53 /50.00 @:0 B@:127
2023-11-30 12:17:58,895 - Recv:  T:24.45 /0.00 B:30.96 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:58,903 - Recv: echo:busy: processing
2023-11-30 12:17:59,897 - Recv:  T:24.49 /0.00 B:31.40 /50.00 @:0 B@:127 W:?
2023-11-30 12:17:59,906 - Recv:  T:24.49 /0.00 B:31.40 /50.00 @:0 B@:127
2023-11-30 12:18:00,894 - Recv:  T:24.50 /0.00 B:31.81 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:00,903 - Recv: echo:busy: processing
2023-11-30 12:18:01,896 - Recv:  T:24.42 /0.00 B:32.29 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:01,905 - Recv:  T:24.42 /0.00 B:32.29 /50.00 @:0 B@:127
2023-11-30 12:18:02,895 - Recv:  T:24.46 /0.00 B:32.78 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:02,904 - Recv: echo:busy: processing
2023-11-30 12:18:03,896 - Recv:  T:24.51 /0.00 B:33.14 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:03,905 - Recv:  T:24.51 /0.00 B:33.14 /50.00 @:0 B@:127
2023-11-30 12:18:04,894 - Recv:  T:24.45 /0.00 B:33.61 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:04,903 - Recv: echo:busy: processing
2023-11-30 12:18:05,896 - Recv:  T:24.48 /0.00 B:34.03 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:05,905 - Recv:  T:24.48 /0.00 B:34.03 /50.00 @:0 B@:127
2023-11-30 12:18:06,894 - Recv:  T:24.45 /0.00 B:34.52 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:06,903 - Recv: echo:busy: processing
2023-11-30 12:18:07,896 - Recv:  T:24.47 /0.00 B:34.88 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:07,905 - Recv:  T:24.47 /0.00 B:34.88 /50.00 @:0 B@:127
2023-11-30 12:18:08,895 - Recv:  T:24.46 /0.00 B:35.28 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:08,904 - Recv: echo:busy: processing
2023-11-30 12:18:09,895 - Recv:  T:24.49 /0.00 B:35.68 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:09,904 - Recv:  T:24.49 /0.00 B:35.68 /50.00 @:0 B@:127
2023-11-30 12:18:10,894 - Recv:  T:24.45 /0.00 B:36.03 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:10,903 - Recv: echo:busy: processing
2023-11-30 12:18:11,896 - Recv:  T:24.44 /0.00 B:36.30 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:11,905 - Recv:  T:24.44 /0.00 B:36.30 /50.00 @:0 B@:127
2023-11-30 12:18:12,894 - Recv:  T:24.42 /0.00 B:36.61 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:12,903 - Recv: echo:busy: processing
2023-11-30 12:18:13,896 - Recv:  T:24.44 /0.00 B:37.01 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:13,905 - Recv:  T:24.44 /0.00 B:37.01 /50.00 @:0 B@:127
2023-11-30 12:18:14,894 - Recv:  T:24.48 /0.00 B:37.29 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:14,903 - Recv: echo:busy: processing
2023-11-30 12:18:15,895 - Recv:  T:24.44 /0.00 B:37.59 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:15,904 - Recv:  T:24.44 /0.00 B:37.59 /50.00 @:0 B@:127
2023-11-30 12:18:16,894 - Recv:  T:24.47 /0.00 B:37.94 /50.00 @:0 B@:127 W:?
2023-11-30 12:18:16,903 - Recv: echo:busy: processing
2023-11-30 12:18:17,066 - Recv: //action:cancel
2023-11-30 12:18:17,067 - Cancelling on request of the printer...
2023-11-30 12:18:17,071 - Changing monitoring state from "Printing" to "Cancelling"
2023-11-30 12:18:17,095 - Send: N5 M108*47
2023-11-30 12:18:17,461 - Recv: //action:prompt_end
2023-11-30 12:18:17,465 - Recv: //action:prompt_begin UI Aborted
2023-11-30 12:18:17,475 - Recv: //action:prompt_button Dismiss
2023-11-30 12:18:17,485 - Recv: //action:prompt_show
2023-11-30 12:18:17,491 - Recv: ok N4 P15 B15
2023-11-30 12:18:17,494 - Recv: //action:notification Print Aborted

Moosiecakes, 2023-11-30:
I'm going to try a different Micro USB cable. I'm wondering if that's an issue.  I do have a piece of Electrical Tape over the Power pin on the USB.

mriscoc, 2023-12-01:
From the logs, the printer is requesting a cancel action.
```
//action:cancel
```
But there is no information about why the printer cancels the print job.

Moosiecakes, 2023-12-01:
> From the logs, the printer is requesting a cancel action.
> 
> ```
> //action:cancel
> ```
> 
> But there is no information about why the printer cancels the print job.

@mriscoc Sorry, I cancelled the job on the printer when i saw it had done it again. I thought perhaps the logs would have shown some error on the initial startup. 

I can certainly reproduce it again and leave it until it starts going around the build plate not extruding.