mriscoc/Ender3V2S1 #1162: [BUG] Filament run-out sensor no longer works

gexgecko, 2023-12-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Had a print run out of filament, the LED in the sensor went out, but the print did not pause.

### Bug Timeline

some rev AFTER 20230312

### Expected behavior

Print stops, and prompts for more filament.

### Actual behavior

Print continues, and just moves around mid-air with no filament

### Steps to Reproduce

Run a print, trip the filament run-out sensor

### Version of Professional Firmware

20230904

### Printer model

Ender3 S1

### Electronics

4.2.2

### Add-ons

None

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-12-18:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

mriscoc, 2023-12-18:
There is no precompiled binary for this combination:  
  
[image]

gexgecko, 2023-12-18:
Hello, perhaps I don't know how to align the Electronics version (4.2.2) with the board rev then.
I am using:
"Ender3S1-F1-UBL-MPC-20230904.bin"
for my STM32F103 Ender3 S1

mriscoc, 2023-12-19:
Please use the support channels posted here: https://github.com/mriscoc/Ender3V2S1/issues/1162#issuecomment-1859588761

closed 2023-12-19 by mriscoc