mriscoc/Ender3V2S1 #1166: [BUG] (New Firmware keeps saying Z is triggered)

OldNRelentless, 2023-12-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Since update it advises the Z is triggered and is not operating the CR touch. The only way it will continue is if the Z stop is reinstalled on the side.

### Bug Timeline

New Today

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231202.bin

### Printer model

Ender 3 V2

### Electronics

4.2.7

### LCD/Controller

4.3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

mriscoc, 2023-12-03:
You can't use a 422 firmware in a 4.2.7 board

mriscoc, 2023-12-03:
This is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-02-12 by github-actions[bot]