mriscoc/Ender3V2S1 #1169: E3 neo V2 prints only "invisible" filament BUG] (bug summary)

atreyusfate, 2023-12-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

since installing the fantastic firmware, got the problem that my E3 Neo V2 prints "invisible" filament only ;) all other features were great except this problem. hotend and heatbreak are fresh cleaned, the Temp of the filament is ok too, maybe its on the extruderv ?? had done an upgrade to the "red alu" extruder...

### Bug Timeline

since installing the first firmware mod yesterday

### Steps to Reproduce

Done aall steps to calibrate alright, were gladly if someone knows what the heckmeck this is ... my Ender got a CR-Touch , bed leveling and z axis were fine

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231202.bin

### Printer model

Ender 3 Neo V2

### Electronics

4.2.2 board

### LCD/Controller

TJC display

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

Cura 5.5 were used, Anycubic Printers pronting well, but the new one here from Crealty flops me above days

atreyusfate, 2023-12-09:
the problem is that i dont can flash the stock firmware....why ever

mriscoc, 2023-12-10:
Check the extruder minimum temperature in Advanced/Filament settings menu.

mriscoc, 2023-12-18:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

closed 2024-02-27 by github-actions[bot]