mriscoc/Ender3V2S1 #1175: [BUG] Disable one of more toolbar options causes screen corruption and reboot if scrolled

Crazy-Charles, 2023-12-16:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Screen corruption and reboot if one or more toolbar items are disable

### Bug Timeline

New 

### Expected behavior

Too continue operating

### Actual behavior

Crashed and rebooted

### Version of Professional Firmware

Ender3V2-427-BLTUBL-MPC-20231208.bin

### Printer model

3V2

### Electronics

4.2.7

### LCD/Controller

Dwin

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

[image]
[image]
[image]

mriscoc, 2023-12-16:
Thank you. Fixed in private development branch, the fix will be public at the next demo release.

mriscoc, 2024-01-08:
Fixed in current sources

closed 2024-01-08 by mriscoc