mriscoc/Ender3V2S1 #1176: [BUG] JPG thumbnail not shown.

Crazy-Charles, 2023-12-16:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

A gcode file embedded with a JPG file has "thumbnail_JPG begin" as a header, Proui library searches for "jpg begin" and as the strings are not equal it does display the thumbnail although it is in a valid format.

### Bug Timeline

Old issue - corrected using a hex editor

### Expected behavior

Gcode preview to display JPG thumbnail

### Actual behavior

Reports "Thumbnail not found"

### Version of Professional Firmware

Ender3V2-427-BLTUBL-MPC-20231208.bin

### Printer model

3V2

### Electronics

4.2.7

### LCD/Controller

Dwin

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

mriscoc, 2023-12-16:
You must use the Professional firmware script to show thumbnails. Check here: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview

Crazy-Charles, 2023-12-16:
Yes i am using that script, the issue is if you choose JPG as the file format rather than PNG in PrusaSlicer the script still produces a valid "base 64" image file but the header in the gcode file is e.g. "; thumbnail_JPG begin 230x180 21824", the Proui library libproui_ubl_dwin.a file at address 0xC3BC has a lower case "jpg", changing this to uppercase allows the file to be processed and the image displayed.

I know this doesn't fully follow your instructions but i do have quite a few print files that were created before gcode_preview was available with JPG thumbnails and then post processed them, obviously the function is there but not working.

closed 2024-02-25 by github-actions[bot]