mriscoc/Ender3V2S1 #1180: Latest firmware bootloops when compiled from scratch for Ender 3 S1 F4

stickmansam4, 2023-12-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When compiling MRISCO Source from scratch, causes Ender 3 S1 to soft brick and bootloop. 

However if  i load the precompiled binary file from GIT repo "Ender3S1-F4-UBL-20231214.bin" it boots just fine.

### Bug Timeline

New, 2023 Release

### Expected behavior

I expect the newly complied binary to boot successfully just the same as the precompiled binary provided from GIT

### Actual behavior

printer hangs on Creality splash screen and seems to bootloop when viewing the COM port.

### Steps to Reproduce

1) Unpack MRISCO Source from GIT "Ender3V2S1-Ender3V2S1-Released.zip"
2) Unpack MRISCO Configurations from GIT "Special_Configurations-main.zip"
3) Using IDLE, run "Configurator.pyw" 
4) Select Ender 3 S1 stock settings shown attached PNG
5) Generate Files, and move to .../src/Marlin/
6) moved platformio.ini to .../src/
7) Execute Build "STM32F401RC (256K)" 
8) Create "SMT32F4_UPDATE" folder on root of SDCARD and copy .bin file into that folder. 
9) insert into printer and power on. 

### Version of Professional Firmware

2.1.3

### Printer model

Creality Ender 3 S1, F4

### Electronics

STM23F401RC

### LCD/Controller

DWIN

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[BuildLog.txt](https://github.com/mriscoc/Ender3V2S1/files/13711427/BuildLog.txt)
[image]
[image]

mriscoc, 2023-12-19:
Thanks, that issue was reported first here: https://github.com/mriscoc/Ender3V2S1/discussions/1164#discussioncomment-7805341

Indeed the current sources and precompiled firmware for the F4 202312XX version have a bug that prevents to boot. Unfortunately I don't have that board for do the proper debug. I suspect about the CRC corruption in settings.cpp.

I need the debugging information sent by the printer to a serial console when the printer boots.

stickmansam4, 2023-12-19:
Ok I will try to get that info tonight.

I tried several configurations and even manually disabled the items you mentioned in your post to reduce program size but still gets the same result.

Last time I hooked the serial console up during boot it read back mostly garbage characters but eventually I started to see repeating "SD bootloader" and then the name of the bin file over and over.

That was using baud 115200, all other baudrates were illegible.

thecactiboi, 2023-12-19:
I have experienced this same or a similar issue with my F4 S1, when trying to load a custom-compiled professional firmware from SD card the printer is stuck on the "wait for reboot" screen and stays like this until another firmware is flashed. When connected to a host (pronterface at 250000 baud), this is the output I received:

> Marlin ProUI Version: 2.1.3 MRiscoC
> CUSTOM BUILD: 20231214 - 13:42
> Author: Miguel Risco-Castillo (MRiscoC)
> echo: Free Memory: 42351  PlannerBufferBytes: 1728
> Warning:EEPROM CRC mismatch - (stored) 62963 != 64046 (calculated)!
> //action:notification Err: EEPROM CRC
> //action:notification Hardcoded Default Settings Loaded
> echo:Hardcoded Default Settings Loaded
> echo:Settings Stored (1065 bytes; crc 62963)
> //action:notification Settings Stored
> echo:EEPROM Initialized
> Got rubbish reply from COM3 at baudrate 250000:
> Maybe a bad baudrate?
>  Software Reset

This repeated until the printer was turned off, and rebooting even without the SD card inserted produced the same output.

By disabling EEPROM_AUTO_INIT and EEPROM_INIT_NOW and adding DEBUG_EEPROM_READWRITE in Configuration.h and recompiling I was able to get the custom firmware to boot, though it would give the message "Error: EEPROM CRC" on the printer screen at startup and when rebooted would discard any saved settings. Sometimes a message with "EEPROM Size" would show on the printer screen as well.

mriscoc, 2023-12-20:
Thanks, that confirms my suspicions, CRC was corrupt. So, a firmware where CRC is tested in different places is needed to discover where is the CRC corrupted, please enable DEBUG_EEPROM_READWRITE and paste here the serial console results.

stickmansam4, 2023-12-20:
Just got home and tried it as well looks like i got the same thing.
#
Marlin ProUI Version: 2.1.3 MRiscoC
CUSTOM BUILD: 20231218 - 20:18
Author: Miguel Risco-Castillo (MRiscoC)
echo: Free Memory: 53711  PlannerBufferBytes: 1856
Warning:EEPROM CRC mismatch - (stored) 31692 != 34548 (calculated)!
//action:notification Err: EEPROM CRC
//action:notification Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
echo:Settings Stored (1050 bytes; crc 31692)
//action:notification Settings Stored
echo:EEPROM Initialized
#

thecactiboi, 2023-12-20:
Here is the output I received when DEBUG_EEPROM_READWRITE was enabled and EEPROM_BOOT_SILENT was disabled:

>  External Reset
> Marlin ProUI Version: 2.1.3 MRiscoC
> CUSTOM BUILD: 20231219 - 21:47
> Author: Miguel Risco-Castillo (MRiscoC)
> echo: Free Memory: 54431  PlannerBufferBytes: 1536
> echo:EEPROM version mismatch (EEPROM=V89 Marlin=P89)
> //action:notification Hardcoded Default Settings Loaded
> echo:Hardcoded Default Settings Loaded
> Field: proui_data
> Field: e_factors
> Field: home_offset
> Field: runout_sensor_enabled
> Field: planner_autotemp_max
> Field: probe_offset
> Field: planner_leveling_active
> Field: servo_angles
> Field: bltouch_od_5v_mode
> Field: bltouch_high_speed_mode
> Field: ui_material_preset
> Field: hotendPID
> Field: lpq_len
> Field: bedPID
> Field: chamberPID
> Field: power_monitor_flags
> Field: lcd_contrast
> Field: lcd_brightness
> Field: controllerFan_settings
> Field: recovery_enabled
> Field: fwretract_settings
> Field: parser_volumetric_enabled
> Field: tmc_stepper_current
> Field: tmc_hybrid_threshold
> Field: tmc_stealth_enabled
> Field: planner_extruder_advance_K
> Field: motor_current_setting
> Field: coordinate_system
> Field: fc_settings
> Field: backlash_distance_mm
> Field: dwin_data
> echo:Settings Stored (1049 bytes; crc 7352)
> //action:notification Settings Stored
> echo:EEPROM Initialized
> Got rubbish reply from COM3 at baudrate 250000:
> Maybe a bad baudrate?
>  Software Reset
> Marlin ProUI Version: 2.1.3 MRiscoC
> CUSTOM BUILD: 20231219 - 21:47
> Author: Miguel Risco-Castillo (MRiscoC)
> echo: Free Memory: 54431  PlannerBufferBytes: 1536
> Field: proui_data
> Field: e_factors
> Field: home_offset
> Field: runout_sensor_enabled
> Field: probe_offset
> Field: planner_leveling_active
> Field: servo_angles
> Field: bltouch_od_5v_mode
> Field: bltouch_high_speed_mode
> Field: ui_material_preset
> Field: lpq_len
> Field: power_monitor_flags
> Field: lcd_contrast
> Field: lcd_brightness
> Field: controllerFan_settings
> Field: recovery_enabled
> Field: fwretract_settings
> Field: parser_volumetric_enabled
> Field: tmc_stepper_current
> Field: tmc_hybrid_threshold
> Field: tmc_sgt
> Field: tmc_stealth_enabled
> Field: planner_extruder_advance_K
> Field: motor_current_setting
> Field: coordinate_system
> Field: fc_settings
> Field: backlash_distance_mm
> Field: dwin_data
> Warning:EEPROM CRC mismatch - (stored) 7352 != 10178 (calculated)!
> //action:notification Err: EEPROM CRC
> //action:notification Hardcoded Default Settings Loaded
> echo:Hardcoded Default Settings Loaded
> Field: proui_data
> Field: e_factors
> Field: home_offset
> Field: runout_sensor_enabled
> Field: planner_autotemp_max
> Field: probe_offset
> Field: planner_leveling_active
> Field: servo_angles
> Field: bltouch_od_5v_mode
> Field: bltouch_high_speed_mode
> Field: ui_material_preset
> Field: hotendPID
> Field: lpq_len
> Field: bedPID
> Field: chamberPID
> Field: power_monitor_flags
> Field: lcd_contrast
> Field: lcd_brightness
> Field: controllerFan_settings
> Field: recovery_enabled
> Field: fwretract_settings
> Field: parser_volumetric_enabled
> Field: tmc_stepper_current
> Field: tmc_hybrid_threshold
> Field: tmc_stealth_enabled
> Field: planner_extruder_advance_K
> Field: motor_current_setting
> Field: coordinate_system
> Field: fc_settings
> Field: backlash_distance_mm
> Field: dwin_data
> echo:Settings Stored (1049 bytes; crc 7352)
> //action:notification Settings Stored
> echo:EEPROM Initialized
> Got rubbish reply from COM3 at baudrate 250000:
> Maybe a bad baudrate?
>  Software Reset

When EEPROM_AUTO_INIT and EEPROM_INIT_NOW were disabled, this is what I got on boot:

>   External Reset
> Marlin ProUI Version: 2.1.3 MRiscoC
> CUSTOM BUILD: 20231219 - 22:00
> Author: Miguel Risco-Castillo (MRiscoC)
> echo: Free Memory: 54431  PlannerBufferBytes: 1536
> Warning:EEPROM datasize error.
> echo:Index: 8 Size: 1045
> //action:notification Err: EEPROM Size
> //action:notification Hardcoded Default Settings Loaded
> echo:Hardcoded Default Settings Loaded
> //action:prompt_end
> BL24CXX Check succeeded!
> echo:SD card ok
> Printer is now online.

I then clicked the "Store Settings" option on the printer, which slowly output these:

> Field: proui_data
> Field: e_factors
> Field: home_offset
> Field: runout_sensor_enabled
> Field: planner_autotemp_max
> Field: probe_offset
> Field: planner_leveling_active
> Field: servo_angles
> Field: bltouch_od_5v_mode
> Field: bltouch_high_speed_mode
> Field: ui_material_preset
> Field: hotendPID
> Field: lpq_len
> Field: bedPID
> Field: chamberPID
> Field: power_monitor_flags
> Field: lcd_contrast
> Field: lcd_brightness
> Field: controllerFan_settings
> Field: recovery_enabled
> Field: fwretract_settings
> Field: parser_volumetric_enabled
> Field: tmc_stepper_current
> Field: tmc_hybrid_threshold
> Field: tmc_stealth_enabled
> Field: planner_extruder_advance_K
> Field: motor_current_setting
> Field: coordinate_system
> Field: fc_settings
> Field: backlash_distance_mm
> Field: dwin_data
> Got rubbish reply from COM3 at baudrate 250000:
> Maybe a bad baudrate?

At this point the printer froze and quickly rebooted. When the reboot finished, Err: EEPROM Version showed on the printer's status bar. On subsequent reboots this does not happen when saving settings, though settings are still not saved when turning the printer off and on.

classicrocker883, 2023-12-20:
>On subsequent reboots this does not happen when saving settings, though settings are still not saved when turning the printer off and on.

>echo:EEPROM version mismatch (EEPROM=V89 Marlin=P89)

that is also something I have experience. but only using Bilinear_auto_bed_leveling.

Arturchik29, 2023-12-22:
It looks like the problems are with these settings:
motor_current_setting
tmc_sgt

Quick fix: comment out everything about these settings in Marlin\src\module\settings.cpp
Also define new eeprom structure vesion here:
#define EEPROM_VERSION "P89"
change to
#define EEPROM_VERSION "P90"

mriscoc, 2023-12-23:
> It looks like the problems are with these settings: motor_current_setting tmc_sgt
> 
> Quick fix: comment out everything about these settings in Marlin\src\module\settings.cpp Also define new eeprom structure vesion here: #define EEPROM_VERSION "P89" change to #define EEPROM_VERSION "P90"

Thanks for your findings, but I can't understand why F4 UBL version doesn't suffer of bootloops. Trying to give more details, if follow your steps. there isn't a bootloop, but ¿are the settings saved and loaded correctly?

thecactiboi, 2023-12-24:
I made the changes suggested by @Arturchik29, commenting out all instances of motor_current_setting and tmc_sgt in settings.cpp, as well as changing the EEPROM version to P90. At least in my case, the printer _was_ able to save and load settings on reboot when I made these changes to my custom firmware build. I also tried only changing the EEPROM version, but it went back to bootlooping as usual, so there appears to be something wrong with motor_current_setting and tmc_sgt being present in settings.cpp.

Arturchik29, 2023-12-24:
> Thanks for your findings, but I can't understand why F4 UBL version doesn't suffer of bootloops

[image]

This simple configuration will not compile due to flash overflow. Could this be due to differences in dependency library versions? Build log:

```
Processing STM32F401RC_creality (board: genericSTM32F401RC; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Processing Professional Firmware requirements
STM32F4 Architecture detected     
DWIN UI detected
Unified Mesh Bed Leveling detected
ProUI detected
Copying variant MARLIN_CREALITY_STM32F401 to framework directory...
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F401RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F401RC (64k RAM. 256k Flash)
HARDWARE: STM32F401RC 84MHz, 64KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\HAL.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\HAL_SPI.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\MarlinSPI.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\MarlinSerial.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\MinSerial.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\Servo.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_bl24cxx.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_flash.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_if_iic.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_sdcard.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_sram.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\eeprom_wired.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\fast_pwm.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\fastio.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\msc_sd.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\sdio.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\timers.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\usb_host.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\STM32\usb_serial.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\MarlinCore.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\core\serial.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\core\utility.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\feature\babystep.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\feature\bedlevel\bedlevel_tools.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\feature\bedlevel\ubl\ubl.cpp.o
[... 124 lines cut ...]
Compiling .pio\build\STM32F401RC_creality\src\src\module\endstops.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\motion.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\planner.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\printcounter.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\probe.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\servo.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\settings.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\stepper.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\stepper\indirection.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\temperature.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\module\tool_change.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\prouiex\custom_gcodes.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\Sd2Card.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\SdBaseFile.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\SdFatUtil.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\SdFile.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\SdVolume.cpp.o
Compiling .pio\build\STM32F401RC_creality\src\src\sd\cardreader.cpp.o
[... cut ...]
c:/users/workbench/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F401RC_creality/firmware.elf section `.rodata' will not fit in region `FLASH'
c:/users/workbench/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 9948 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F401RC_creality\firmware.elf] Error 1
============================================================= [FAILED] Took 183.45 seconds =============================================================
```

Arturchik29, 2023-12-24:
> are the settings saved and loaded correctly?

yes

mriscoc, 2023-12-24:
Please test it:
[firmware-20231224-140101.zip](https://github.com/mriscoc/Ender3V2S1/files/13762632/firmware-20231224-140101.zip)

Arturchik29, 2023-12-24:
It looks like something happened with the bootloader. I cant reflash to custom build or stock firmware. It's always "OFFICIAL BUILD: 20231224 - 14:00"
There is also an empty DWIN_SET folder in STM32F4_UPDATE. Never seen this before. It is now created by the bootloader every time on strartup if STM32F4_UPDATE exists

mriscoc, 2023-12-24:
> It looks like something happened with the bootloader. I cant reflash to custom build or stock firmware. It's always "OFFICIAL BUILD: 20231224 - 14:00" There is also an empty DWIN_SET folder in [...]

Try with another SD card, format it.

mriscoc, 2023-12-26:
> This simple configuration will not compile due to flash overflow. Could this be due to differences in dependency library versions? Build log:

I think that was covered in another issue, it is a matter of lack of program memory in the STM32F4xxx SoC.

mriscoc, 2024-01-08:
This firmware build should help us to know where the CRC was corrupted, please log the error messages 
[Ender3S1-F4-DEBUG-20240108.zip](https://github.com/mriscoc/Ender3V2S1/files/13865457/Ender3S1-F4-DEBUG-20240108.zip)

Witska, 2024-01-10:
> This firmware build should help us to know where the CRC was corrupted, please log the error messages for EEPROM read and EEPROM write without change any settings. [Ender3S1-F4-DEBUG-20240108.zip](https://github.com/mriscoc/Ender3V2S1/files/13865457/Ender3S1-F4-DEBUG-20240108.zip)

 External Reset
Marlin ProUI Version: 2.1.3 MRiscoC
OFFICIAL BUILD: 20240108 - 14:18
Author: Miguel Risco-Castillo (MRiscoC)
echo: Free Memory: 53719  PlannerBufferBytes: 1856
READ: stored_ver CRC: 18057
READ: stored_hash CRC: 32557
READ: stored_crc CRC: 5859
READ: stored_size CRC: 44129
READ: meshSet CRC: 9031
Field: proui_data
READ: proui_data CRC: 29806
READ: meshViewer.meshmode CRC: 16434
Field: e_factors
READ: e_factors CRC: 27365
READ: (uint8_t *)tmp1 CRC: 41944
READ: planner.settings.min_segment_time_us CRC: 35965
READ: (uint8_t *)tmp2 CRC: 63355
READ: (uint8_t *)tmp3 CRC: 59653
READ: planner.settings.acceleration CRC: 6336
READ: planner.settings.retract_acceleration CRC: 13494
READ: planner.settings.travel_acceleration CRC: 17813
READ: planner.settings.min_feedrate_mm_s CRC: 16656
READ: planner.settings.min_travel_feedrate_mm_s CRC: 23371
READ: planner.max_jerk CRC: 23743
READ: dummyf CRC: 57242
Field: runout_sensor_enabled
READ: runout_sensor_enabled CRC: 41106
READ: runout_distance_mm CRC: 237
READ: new_z_fade_height CRC: 48227
READ: planner.autotemp.max CRC: 64431
READ: planner.autotemp.min CRC: 63611
READ: planner.autotemp.factor CRC: 24324
Field: probe_offset
READ: zpo CRC: 53067
READ: grid_max_x CRC: 62090
READ: grid_max_y CRC: 54388
READ: grid_check CRC: 36584
READ: spacing CRC: 59614
READ: start CRC: 7429
READ: bedlevel.z_values CRC: 59994
Field: bltouch_od_5v_mode
READ: bltouch_od_5v_mode CRC: 1636
Field: bltouch_high_speed_mode
READ: bltouch_high_speed_mode CRC: 5351
Field: ui_material_preset
READ: ui.material_preset CRC: 45534
READ: pidcf CRC: 22774
Field: lpq_len
READ: lpq_len CRC: 51064
READ: pid CRC: 62587
Field: lcd_brightness
READ: lcd_brightness CRC: 23523
READ: ui.backlight_timeout_minutes CRC: 22587
Field: recovery_enabled
READ: recovery_enabled CRC: 57597
Field: fwretract_settings
READ: fwretract_settings CRC: 63582
READ: autoretract_enabled CRC: 12311
Field: parser_volumetric_enabled
READ: storage CRC: 60545
Field: coordinate_system
Field: fc_settings
READ: fc_settings CRC: 34157
Field: dwin_data
READ: dwin_data CRC: 24735
Field: sound_on
READ: ui.sound_on CRC: 58247
READ: hotend_idle.cfg CRC: 47205
Warning:EEPROM CRC mismatch - (stored) 55430 != 47205 (calculated)!
//action:notification Err: EEPROM CRC
//action:notification Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
WRITE: ver CRC: 35498
WRITE: data_size CRC: 44129
WRITE: meshSet CRC: 9031
Field: proui_data
WRITE: proui_data CRC: 29806
WRITE: meshViewer.meshmode CRC: 16434
Field: e_factors
WRITE: e_factors CRC: 27365
WRITE: planner.settings CRC: 23371
WRITE: planner.max_jerk CRC: 23743
WRITE: dummyf CRC: 57242
Field: runout_sensor_enabled
WRITE: runout_sensor_enabled CRC: 41106
WRITE: runout_distance_mm CRC: 237
WRITE: zfh CRC: 48227
Field: planner_autotemp_max
WRITE: planner.autotemp.max CRC: 64431
WRITE: planner.autotemp.min CRC: 63611
WRITE: planner.autotemp.factor CRC: 24324
Field: probe_offset
WRITE: zpo CRC: 53067
WRITE: grid_max_x CRC: 62090
WRITE: grid_max_y CRC: 54388
WRITE: grid_check CRC: 36584
WRITE: bedlevel.grid_spacing CRC: 59614
WRITE: bedlevel.grid_start CRC: 7429
WRITE: bedlevel.z_values CRC: 59994
Field: bltouch_od_5v_mode
WRITE: bltouch_od_5v_mode CRC: 1636
Field: bltouch_high_speed_mode
WRITE: bltouch_high_speed_mode CRC: 5351
Field: ui_material_preset
WRITE: ui.material_preset CRC: 45534
Field: hotendPID
WRITE: pidcf CRC: 22774
Field: lpq_len
WRITE: lpq_len CRC: 51064
Field: bedPID
WRITE: bed_pid CRC: 62587
Field: lcd_brightness
WRITE: lcd_brightness CRC: 23523
WRITE: ui.backlight_timeout_minutes CRC: 22587
Field: recovery_enabled
WRITE: recovery_enabled CRC: 57597
Field: fwretract_settings
WRITE: fwretract.settings CRC: 63582
WRITE: autoretract_enabled CRC: 12311
Field: parser_volumetric_enabled
WRITE: parser.volumetric_enabled CRC: 8531
WRITE: planner.filament_size CRC: 37226
WRITE: dummyf CRC: 60545
Field: coordinate_system
Field: fc_settings
WRITE: fc_settings CRC: 34157
Field: dwin_data
WRITE: dwin_data CRC: 24735
WRITE: ui.sound_on CRC: 58247
WRITE: hotend_idle.cfg CRC: 55430
WRITE: version CRC: 58035
WRITE: build_hash CRC: 24014
WRITE: final_crc CRC: 47914
echo:Settings Stored (823 bytes; crc 55430)
//action:notification Settings Stored
echo:EEPROM Initialized
//action:prompt_end
BL24CXX Check succeeded!
echo:SD card ok

mriscoc, 2024-01-12:
Excellent:
```
READ: hotend_idle.cfg CRC: 47205
WRITE: hotend_idle.cfg CRC: 55430
```
So, now I need to analyze why the CRC is different.

dmenzie, 2024-01-18:
Any update to this? i suspect im running into the same issue with my compiled build but am not tech savvy enough to be useful for debugging

dmenzie, 2024-01-18:
> It looks like something happened with the bootloader. I cant reflash to custom build or stock firmware. It's always "OFFICIAL BUILD: 20231224 - 14:00" There is also an empty DWIN_SET folder in [...]

i am also unable to reflash after using this firmware

Arturchik29, 2024-01-20:
@dmenzie unfortunately, I my case the only thing what can help is reflashing with ST-Link. The bootloader area seems to be overwritten after flashing. Do you need help with this?

Arturchik29, 2024-01-20:
> Excellent:
> 
> ```
> READ: hotend_idle.cfg CRC: 47205
> WRITE: hotend_idle.cfg CRC: 55430
> ```
> 
> So, now I need to analyze why the CRC is different.

Can you upload your test builds for my configuration?
[image]

Previous builds cause to bootloader corruption and I don't know why. Can this be due to the display version (my display TJC)
Can you share the start of your build output log? There are lot of differences between my build and yours. For example if I set UBL check in configurator, firmware wont compile at all (even if no features selected).
Let's make shure we use the same libs. It may help solve the issue:

```
Processing STM32F401RC_creality (board: genericSTM32F401RC; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Processing Professional Firmware requirements
STM32F4 Architecture detected  
DWIN UI detected
Auto Mesh Bed Leveling detected
ProUI detected
Copying variant MARLIN_CREALITY_STM32F401 to framework directory...
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F401RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F401RC (64k RAM. 256k Flash)
HARDWARE: STM32F401RC 84MHz, 64KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0   
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1      
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2       
Building in release mode

```

mriscoc, 2024-01-20:
@Arturchik29  that is a different issue already answered, check the issue history.

dmenzie, 2024-01-21:
@Arturchik29 yes please, any advice/guidance would be appreciated. I have an ST-link clone and/or an Arduino to get it done, but am unfamiliar with programming microcontrollers, id assume the process is similar to an older ender 3 (which I can find video documentation on how to do), but am unsure if there is anything specific to the stm32f4 boards that I need to be aware of

mriscoc, 2024-01-23:
Anyone who wants to help resolve this issue can join us here: https://t.me/ender3v2s1firmware/66444

mriscoc, 2024-01-24:
Issue solved!, We just found that Creality implemented some kind of flag zone in the EEPROM for the Ender3S1 F4 bootloader, if that zone is written with non 0 values the bootloader refuses to flash a new firmware.

That solves the mystery of why that printer sometimes refuses to update new firmware and only can be solved by using a ST-link device, also that solves the mystery about the CRC EEPROM corruption

TheWebMachine, 2024-01-24:
That also means we have even less space to work with. 😮‍💨

mriscoc, 2024-01-24:
> That also means we have even less space to work with. 😮‍💨

it is not a problem with the flash program memory is a matter of EEPROM memory

TheWebMachine, 2024-01-24:
With the new per-slot ABL 2.0 settings, are we at any risk of filling it up?

mriscoc, 2024-01-24:
> With the new per-slot ABL 2.0 settings, are we at any risk of filling it up?

UBL has its own EEPROM zone out of that flag zone. But normal firmware without UBL is affected.

TheWebMachine, 2024-01-24:
And we don't have room for UBL in the CVLM firmware without some other major sacrifice, correct?

mriscoc, 2024-01-24:
> And we don't have room for UBL in the CVLM firmware without some other major sacrifice, correct?

But don't worry, the solution is on the way, we are trying it here: https://t.me/ender3v2s1firmware/66444

TheWebMachine, 2024-01-24:
I could perhaps give it a try this evening.

mriscoc, 2024-01-25:
A PR https://github.com/MarlinFirmware/Marlin/pull/26729 was posted in Marlin repository which solves this issue.

mriscoc, 2024-01-25:
Resolved in the latest sources.

closed 2024-01-25 by mriscoc

mriscoc, 2024-01-26:
For people with an Ender3S1F4 printer which refuses to flash new firmware @trotus has prepared a G-code file which could help with the flash process, simply put the new firmware and this file in the SD card and select it.

```gcode
; Professional Firmware Configuration File
;=====================================================
C10                    ; Mark as a configuration file
M117 Applying configuration
;-----------------------------------------------------
M86 S0 T3 E0 B0        ; Reset flag zone
M500                   ; Save all settings
;-----------------------------------------------------
G4 S1                  ; Wait a second
M300 P200              ; Beep
M117 Configuration Applied
;-----------------------------------------------------
```
[config_reset.zip](https://github.com/mriscoc/Ender3V2S1/files/14060786/config_reset.zip)

trotus47, 2024-01-26:
Yep, small addon. Put this file on SD and new firmware to STM32F4_UPDATE. Select file like in case of printing, when settings will be applied, just reboot

MagicAlien, 2024-02-09:
> For people with an Ender3S1F4 printer which refuses to flash new firmware, @trotus47 has prepared a G-code file which could help with the flash process, simply put the new firmware (in the folder STM32F4_UPDATE), this file in the SD card and select and click on it in the Media menu, then reboot the printer.
> 
> ```gcode
> ; Professional Firmware Configuration File
> ;=====================================================
> C10                    ; Mark as a configuration file
> M117 Applying configuration
> ;-----------------------------------------------------
> M86 S0 T3 E0 B0        ; Reset flag zone
> M500                   ; Save all settings
> ;-----------------------------------------------------
> G4 S1                  ; Wait a second
> M300 P200              ; Beep
> M117 Configuration Applied
> ;-----------------------------------------------------
> ```
> 
> [config_reset.zip](https://github.com/mriscoc/Ender3V2S1/files/14060786/config_reset.zip)

 - The config reset g-code worked well;
 - The compiled firmware from the updated released branch, with Fix Ender-3 S1F4 EEPROM exclusion zone, still doesn't work:
 --> At boot the new firmware is installed as expected;
 --> The printer start with the new boot;
 --> Bootloop rebooting without SD card;
 --> Booting with SD card will re-install the firmware again (without name changing)

mriscoc, 2024-02-09:
> * The compiled firmware from the updated released branch, with Fix Ender-3 S1F4 EEPROM exclusion zone, still doesn't work

None of the S1 F4? or an specific bin file?

MagicAlien, 2024-02-11:
> > * The compiled firmware from the updated released branch, with Fix Ender-3 S1F4 EEPROM exclusion zone, still doesn't work
> 
> None of the S1 F4? or an specific bin file?

I was following the topic, meaning that compiling from scratch the firmware with the "Fix Ender-3 S1F4 EEPROM exclusion zone" won't work.
To fix properly the EEPROM exclusion zone, I had to #define EEPROM_EXCL_ZONE **915**,926 [was 916]

mriscoc, 2024-02-12:
In our test board the 916 was the start of the EEPROM exclusion zone that was checked doing a EEPROM dump. If the zone is filled with 0xFF values, the boot loader changes it when the printer is rebooted:

```log
0370: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0380: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0390: FF FF FF FF 00 FF FF FF FF FF FF FF FF 66 FF FF
03A0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
03B0: FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
```

So maybe the zone changes depending on the bootloader version installed in the printer board. If more reports are obtained, the exclusion zone will have to be expanded.