mriscoc/Ender3V2S1 #1185: Data loss after shutdown

Slavikpd, 2023-12-22:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

After turning off the printer, the data, Mesh Insert, Mesh Leveling, Mesh Settings will be lost. I enter my data in these items, and after turning it off and on, the data changes to the original, entered in the firmware. It does this since "Ender3V2-422-BLTUBL-MPC-20231202.bin" "Ender3V2-422-BLTUBL-MPC-20231217.bin".

### Bug Timeline

Ender3V2-422-BLTUBL-MPC-20231202

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231202

### Printer model

Ender3V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

TheWebMachine, 2023-12-22:
I had the same issue with the F4 UBL version of the firmware when I tested the 20231202 demo.

josederas, 2023-12-22:
No issues here, but you do need to recall the saved mesh with an M420 L0.  Then the stored meshed is reloaded and can be used.  Other settings such as z-offset, etc. are not lost for me.

Slavikpd, 2023-12-23:
Thanks, I'll try

mriscoc, 2023-12-24:
Due to changes to support AML 2.0, starting with the Anniversary 2023 release in UBL versions, all mesh related data: XY density, inset and leveling data are now saved in slots, so it is now possible to have different densities, insets and data for different bed or printer conditions.

You can use G29 Sn to save and G29 Ln to load UBL mesh data (where n = 0..4)

josederas, 2023-12-24:
Miguel for me the saved slot is not loaded (as OP mentions).  I do a M420 L0 to recall then all good.  At power up, there's no mesh loaded as verified doing an M420 V.

Slavikpd, 2023-12-25:
I uploaded the firmware again. I set the parameters, started Leveling. Saved in the slot. Then after turning the printer off and on, everything was OK. Thank you for advice. With this, I consider the ticket closed.

closed 2023-12-25 by Slavikpd