mriscoc/Ender3V2S1 #1186: Out of bounds error when trying to do tramming wizard

megames07, 2023-12-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I wanted to use the tramming wizard but i get an out of bounds error after the printer hits the first point. Is this a bug or is it something wrong on my side. I dont know how to fix it because my bltouch is atached like it should be. Im using a glass bed but i dont think thats the problem.

### Expected behavior

I thought it would just do the wizard.

### Actual behavior

It stopped at the first point

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20231217.bin

### Printer model

Creality Ender 3v2 Neo

### Electronics

board 4.2.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

Hufpul, 2023-12-24:
Hey, also having the exact same problem. Bed tramming worked in a previous version for me.

Mdigital23, 2023-12-25:
Same here having the issues after flashing the current version.  Bed tramming worked in a previous version for me too. Idk what to  do, I can't recall what version I was using previously.

mriscoc, 2023-12-25:
The Anniversary 2023 firmware uses other structure to save the mesh data, in special for the UBL versions, all mesh related data: XY density, inset and leveling data are now saved in slots, so it is now possible to have different densities, insets and data for different bed or printer conditions.

There is a little algorithm to try to detect if the mesh inset are wrong at boot but due to program space restriction, the test is very basic. So, it is possible that at the first boot you need to set a proper mesh inset following the calibration guide. Then, you can use G29 Sn to save and G29 Ln to load UBL mesh data (where n is the preferred mesh slot), save the slot to be used with M500 so the latest saved slot is automatically loaded at power on.

Mdigital23, 2023-12-28:
> The Anniversary 2023 firmware uses other structure to save the mesh data, in special for the UBL versions, all mesh related data: XY density, inset and leveling data are now saved in slots, so it is now possible to have different densities, insets and data for different bed or printing conditions.
> 
> There is a little algorithm to try to detect if the mesh inset are wrong at boot but due to program space restriction, the test is very basic. So, it is possible that at the first boot you need to set a proper mesh inset following the [calibration guide](https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides). Then, you can use `G29 Sn` to save and `G29 Ln` to load UBL mesh data (where n is the preferred mesh slot), save the slot to be used with M500 so the latest saved slot is automatically loaded at power on.
> 
> Related to #1185

Thank you for your swift response, I've still been having an issue trying to follow the calibration guide, are there any plans for an update that keeps the ease of the tramming wizard like in the past versions of the firmware. Once again thank you for your time, effort, and hard work. Keep it up!

Mdigital23, 2023-12-28:
> The Anniversary 2023 firmware uses other structure to save the mesh data, in special for the UBL versions, all mesh related data: XY density, inset and leveling data are now saved in slots, so it is [...]

Thank you for your swift response, I've still been having issue trying to follow the calibration guide, are there any plans for an update that keeps the ease of the tramming wizard like in the past versions of the firmware. Once again thank you for your time, effort, and hard work. Keep it up!

Mdigital23, 2023-12-29:
Mini update even after following the [calibration guide](https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides) (despite my printer is a stock Ender 3 V2) I'm still receiving the "Bed Tramming-Tramming Error" when trying to use the Tramming wizard feature. 

Even when I was trying to use Manual bed leveling, I found that I could only move the nozzle to the front two points and the back two points are too low for the nozzle to come up on, and no matter how much I loosened my bed screws the nozzle was way too far away.  I'm not sure what's going on at this point, I didn't have this issue before this update. 
[image]

mriscoc, 2023-12-30:
> I didn't have this issue before this update.

Try a manual bed tramming first, the probe only works under certain limits, if the bed is severely tilted the bed tramming wizard will not work.

Mdigital23, 2024-01-03:
> > I didn't have this issue before this update.
> 
> Try a manual bed tramming first, the probe only works under certain limits, if the bed is severely tilted the bed tramming wizard will not work.

So I tried for the last few days, made sure my bed wasn't tilted, I tried to use manual bed tramming, and found that the print head would remain extremely far away and above the final two corners. I re-flashed the printer again with default settings. Same thing happened again.  

So in the end I ended up going to the last version I found stable on my printer, I went all the way to the spring update, and I found the default settings to allow everything to perfectly work the way it did before the anniversary update. I was able use the manual bed tramming properly with the print head moving where it should be, and as well as the bed tramming wizard worked perfectly. So I'm concluding there's something floating around in the anniversary update, I'm not sure, and I can't help there. but this is where I am. thank you for your hard work, Miguel, hopefully all this gets ironed out soon. Good luck.

Mdigital23, 2024-01-03:
> > I didn't have this issue before this update.
> 
> Try a manual bed tramming first, the probe only works under certain limits, if the bed is severely tilted the bed tramming wizard will not work.

So I tried for the last few days, made sure my bed wasn't tilted, I tried to use manual bed tramming, and found that the print head would remain extremely far away and above the final two corners. I re-flashed the printer again with default settings. Same thing happened again.  

So in the end I ended up going to the last version I found stable on my printer, I went all the way to the spring update, and I found the default settings to allow everything to perfectly work the way it did before the anniversary update. I was able use the manual bed tramming properly with the print head moving where it should be, and as well as the bed tramming wizard worked perfectly. So I'm concluding there's something floating around in the anniversary update, I'm not sure, and I can't help there. but this is where I am. thank you for your hard work, Miguel, hopefully all this gets ironed out soon. good luck.

ouspensky-lab, 2024-01-08:
Try Disabling HS Mode; Advanced > Probe Settings > Uncheck Enable HS Mode

carsona14, 2024-01-10:
Hey, just had this same issue. I returned my CR touch and got a new one, it works now. Hopefully that fixes it for you!

Mdigital23, 2024-01-10:
> Hey, just had this same issue. I returned my CR touch and got a new one, it works now. Hopefully, that fixes it for you!

Unfortunately switching between sensors wasn't my issue either, especially since my bltouch worked with previous versions as well, but at the time I switched it from a spare printer I had and got the same results. So, after going back to an older firmware version, everything was fine again, I even switched it back to the original sensor, and haven't had any issues on the older firmware.

sabasigh, 2024-01-27:
Having the same issue on Ender 3 V2 with CR touch, 4.2.2 board. Used Bed Tramming wizard many times perfectly and i just swapped my nozzle, couple reboots and now it will probe Front left, then say Probe out of bounds.

If i do Manual, it can Probe Left Front/Back but not Right. 

Manually leveling the bed its good.

No specs had been changed in the printer...Wizard had been working great.

Mdigital23, 2024-01-27:
> Having the same issue on Ender 3 V2 with CR touch, 4.2.2 board. Used Bed Tramming wizard many times perfectly and i just swapped my nozzle, couple reboots and now it will probe Front left, then say [...]

Thats very similar to my situation, the only way I solved my issue was going back to an older release of the frimware, thats what i''ve been using for several weeks now.

mriscoc, 2024-01-28:
This: https://github.com/mriscoc/Ender3V2S1/issues/1186#issuecomment-1869064716

dkdkusit, 2024-01-28:
> Thats very similar to my situation, the only way I solved my issue was going back to an older release of the frimware, thats what i''ve been using for several weeks now.

Same behavior for me.
I tried to re-calibrate several times... so I give up.
If I'm right, before the last release I was using the September 2023 version but I can't find it in the repository anymore.
Could anyone tell me where to find the September 4.22 BLTUBL-MPC release? (I'm using an Ender 3 v2 Neo without custom parts)
Thanks.

P.S.: I'm a newbie, very newbie... what do you mean by: "_Then, for UBL you can use G29 Sn to save and G29 Ln to load mesh data (where n is the preferred mesh slot), save the slot to be used with M500 so the latest saved slot is automatically loaded at power on._" ?

Mdigital23, 2024-03-31:
Update I still haven't checked if this is still a problem, has anyone checked? I'm still too afraid to update again.

RaulisArg, 2024-04-09:
Hi. Same issue here. Anyone know where can i find the old Mriscoc releases?

xXDangerous, 2024-05-11:
If it may be of help to anyone: I have an Ender 3 S1 on which I uploaded this optimized Firmware with a custom configuration (only LA enabled due to limited space) and I was getting this "Probe out of bounds" error for the 2 back spots when doing Bed Tramming.

What helped was adjusting the Mesh Inset parameters (Advanced -> Mesh Leveling -> Mesh Inset). I then clicked on "Center Area" and now the bed tramming works. Maximizing the mesh will make the probe go out of bounds, but centering the mesh actually solved this.

Hope this helps.

closed 2024-07-21 by github-actions[bot]