mriscoc/Ender3V2S1 #1187: No response to G29 command

beans-are-gross, 2023-12-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When I call G29 in OctoPrint, no response is received and the printer does not level.

### Bug Timeline

New, today

### Expected behavior

I expect it to begin leveling.

### Actual behavior

It stays in its current position.

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3S1-F4-UBL, based on bugfix-2.1.x (Dec 17 2023 23:52:32)

### Printer model

Creality Ender 3 S1

### Electronics

V2.4 Silent Motherboard with TMC2208 Drivers

### LCD/Controller

Linear Knob Model

### Other add-ons

N/A

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

I have tried this link: [https://marlinfw.org/docs/gcode/G029-abl-linear.html](https://marlinfw.org/docs/gcode/G029-abl-linear.html)

Crazy-Charles, 2023-12-24:
You state above that you loaded the UBL firmware, therefore the command to use is G29 P1

beans-are-gross, 2023-12-24:
That worked, thanks!

closed 2023-12-24 by beans-are-gross