mriscoc/Ender3V2S1 #1188: [BUG] System Reboot

ItzRazorYo, 2023-12-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Every time i try to click on anything in the main menu, the system reboots and im lost. Its for the Ender 3 V2 w/ BTT SKR Mini E3V3 (with BLtouch)

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 V2

### Electronics

SKR Mini

### Other add-ons

BL Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

mriscoc, 2023-12-25:
Hi, this is not a general printer support forum.

This Issue Queue is for firmware bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

Enhanced support in Patreon: https://www.patreon.com/mriscoc

Free community support:
- Github discussions: https://github.com/mriscoc/Ender3V2S1/discussions
- Facebook: https://www.facebook.com/groups/professionalfirmware
- Telegram: https://t.me/ender3v2s1firmware
- Reddit: https://www.reddit.com/r/Ender3v2Firmware
- Discord: https://discord.gg/3dprinters

samiresa, 2023-12-29:
Same exact issue!  I thought I was going crazy, every option I choose the screen reboots.  I have tried flashing the firmware of the screen as well as reflashing the printer again and again.  I am unable to select any option on the screen.

I am using the BigTreeTech cable designed to go from the board to the Ender3V2 screen (correcting the pins reversal issue).

I am using the most up to date stable version of your marlin as well as following the directions of your configurator.

Ender 3V2
Sprite Direct Drive
Bigtreetech SKR Mini e3v3
CR Touch

ItzRazorYo, 2023-12-29:
> Same exact issue! I thought I was going crazy, every option I choose the screen reboots. I have tried flashing the firmware of the screen as well as reflashing the printer again and again. I am [...]

I was able to fix it. It turns out the factory miss labled my cable. Here is what you need to do:
Look on the cable, on both ends there should be a white line on one side, and two on the other. The one with the single line, goes into the display, line facing the notch. The other goes into the board, two lines facing notch.

For me, it was the other way around becuase it got mislabeled, but make sure you have your cable the way i said above, otherwise itll send a wack signal to your mobo causing it to restart

mriscoc, 2023-12-30:
https://github.com/mriscoc/Ender3V2S1/issues/774#issuecomment-1463189494

samiresa, 2023-12-30:
> I was able to fix it. It turns out the factory miss labled my cable. Here is what you need to do: Look on the cable, on both ends there should be a white line on one side, and two on the other. The [...]

Dear sweet jesus you just saved my bank account from buying new things, I was just about to go down the klipper route and be done with this whole thing.  WHO THOUGHT THAT WAS A GOOD IDEA?!?!?  Not to mention that 0% of documentation that I found.

samiresa, 2023-12-30:
> [#774 (comment)](https://github.com/mriscoc/Ender3V2S1/issues/774#issuecomment-1463189494)
> 
> [#131 (comment)](https://github.com/mriscoc/Ender3V2S1/discussions/131#discussioncomment-5270003)

You are wonderful, thank you for such great work.  I shall now go back to compiling all my beautiful firmware.

closed 2023-12-31 by mriscoc