mriscoc/Ender3V2S1 #1190: Minor BUG - Running M504 toggles "Enable Runout" OFF/ON

Crazy-Charles, 2023-12-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Minor issue, running M504 from a terminal will toggle the "Enable Runout" OFF/ON.

### Bug Timeline

Maybe old but confirmed still present in latest firmware

### Expected behavior

Nothing to change when validating eeprom data

### Actual behavior

DWIN "Enable Runout" checkbox toggles OFF/ON
M412  "Filament runout " toggles OFF/ON

### Version of Professional Firmware

Ender3V2-427-BLTUBL-MPC-20231217.bin

### Printer model

3V2

### Electronics

4.2.7

### LCD/Controller

Dwin

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

mriscoc, 2023-12-30:
Marlin calls load function when validate EEPROM contents, currently the load routine overwrites runout enable parameter in a validation process:

https://github.com/MarlinFirmware/Marlin/blob/06dc7f4f526833aaab37725e2e95e68030acb94e/Marlin/src/module/settings.cpp#L1924-L1942

Crazy-Charles, 2023-12-30:
Thanks, that is where i "fixed" it as it was annoying as i have an audible warning along with a blinking red icon background when the sensor gets triggered.
I guess when validating the eeprom it shouldn't do this?

mriscoc, 2023-12-31:
If you disable the filament run-out sensor and save settings, it will kept disabled after validating.

closed 2024-04-11 by github-actions[bot]